ThingPulse/esp8266-oled-ssd1306 #167: Cannot use optimiced Brzo_I2C, undefined reference to `system_get_cpu_freq'

moracabanas, 2018-01-14:
I changed from I2C to Brzo_i2c optimiced. When I try to compile it shows this error:

`ATENCIÓN: la librería arduino_568933 pretende ejecutarse sobre arquitectura(s) (esp8266) y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) (esp32).
Archiving built core (caching) in: C:\Users\etc\AppData\Local\Temp\arduino_cache_442652\core\core_espressif_esp32_lolin32_FlashFreq_80,UploadSpeed_921600_dd21ab0bce255a497044776dfde65b13.a
libraries\arduino_568933\brzo_i2c.c.o:(.iram1.literal+0x2c): undefined reference to `system_get_cpu_freq'

libraries\arduino_568933\brzo_i2c.c.o:(.literal.brzo_i2c_setup+0x14): undefined reference to `system_get_cpu_freq'

libraries\arduino_568933\brzo_i2c.c.o: In function `brzo_i2c_start_transaction':

C:\Users\etc\Arduino\libraries\arduino_568933/brzo_i2c.c:989: undefined reference to `**system_get_cpu_freq**'

libraries\arduino_568933\brzo_i2c.c.o: In function `brzo_i2c_setup':

C:\Users\etc\Arduino\libraries\arduino_568933/brzo_i2c.c:1044: undefined reference to `**system_get_cpu_freq**'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compilando para la tarjeta WEMOS LOLIN32.
`

kjdolan99, 2018-03-17:
This library is only for the esp8266 and will not work on an esp32 board. Use the standard library or switch to the hardware implementation for the esp32 if you need more speed. The esp32 is much faster than the esp8266 and the software driver may be good enough.

moracabanas, 2018-03-19:
Thanks you, I'm very noob at the time. I am using I2C on the examples and it works good. I was trying to use brzo_i2c because it was declared faster than i2c and I missed it's only for ESP8266.

closed 2019-05-17 by marcelstoer