ThingPulse/esp8266-oled-ssd1306 #173: Possible to create custom fonts?

MacSass, 2018-02-09:
Hi,
I really like the library - works great - including the font generator.

One question though:
Would it be possible to create custom fonts in the required format? The font generator "only" provides a fixed set of fonts and options - but I would need an Arial font (like the three included ones) but with a 12 and 14 point size.

Any hints how I could generate those?

Thanks - MacSass

lorol, 2018-02-10:
https://rawgit.com/lorol/esp8266-oled-ssd1306/master/resources/glyphEditor.html
And see the menu upper right corner

MacSass, 2018-02-10:
Hi @lorol,
although this looks interesting, I do not understand how I can make use of it to create the additional Arial font sizes ...
The link in upper right "only" points to the online font generator, which only allows to create fonts from a limited selection of open source fonts.
I need to create add. sizes of Arial font ...

Regards - MacSass

lorol, 2018-02-10:
Alright, see then 
https://github.com/lorol/esp8266-oled-ssd1306/blob/master/resources/FontConverterV3/run_jar_V3.bat
Install a JDK, build the JAR and enjoy :)

MacSass, 2018-02-11:
Hi again @lorol ,
thanks for the hints - got it working - this is really helpful!
Regards - MacSass

closed 2019-05-17 by marcelstoer

JonathanBytes, 2024-09-30:
> Alright, see then https://github.com/lorol/esp8266-oled-ssd1306/blob/master/resources/FontConverterV3/run_jar_V3.bat Install a JDK, build the JAR and enjoy :)

Thanks for the code, works flawlessly!