ThingPulse/esp8266-oled-ssd1306 #174: my oled won't display...

giecuk, 2018-02-10:
hello, what should i change if my oled not display anything, it's not broken oled, because if i used another library it will work.

TD-er, 2018-02-16:
Make sure you don't forget to set these:

```C++
sendCommand(SETVCOMDETECT); //0xDB, (additionally needed to lower the contrast)
sendCommand(0x40);	        //0x40 default, to lower the contrast, put 0
```

On some displays that makes the difference between working and not working.

See also my PR: #171 
That includes about all the PR's and suggestions made here and elsewhere about these displays.

SilverNodashi, 2018-09-28:
> Make sure you don't forget to set these:
>
> ```c++
> sendCommand(SETVCOMDETECT); //0xDB, (additionally needed to lower the contrast)
> sendCommand(0x40);	        //0x40 default, to lower the contrast, put [...]

Where do you set those?

I added it to setup() but got the following error:

'sendCommand' was not declared in this scope

TD-er, 2018-09-28:
I guess a lot may have changed since I added that comment.
Just about all files in the repository have changed since.

Maybe you could share a bit of your code indicating where you call this `sendCommand`?

closed 2019-12-04 by stale[bot]