ThingPulse/esp8266-oled-ssd1306 #176: how to include new created fonts?

tofrnr, 2018-02-13:
hello,
I have created a couple of *.h font files with the font creator 
http://oleddisplay.squix.ch/
 and stored them a subfolder "fonts" in the libraries directory:

#include <fonts\Monospaced_plain_8.h>
#include <fonts\Monospaced_plain_10.h>
#include <fonts\Monospaced_plain_12.h>
#include <fonts\Monospaced_plain_16.h>
#include <fonts\Monospaced_plain_24.h>

unfortunately it does not compile with my example code.
What is the correct way to make it work?

kosso, 2018-02-17:
Try a forward slash? Which might seem odd on Windows, but I think C++ requires it to be forward. 

`#include <fonts/Monospaced_plain_8.h>`

tofrnr, 2018-02-18:
what ??
paths in Windows always are by backslash !!
e.g.:
// Wifi data from  "data\settings.h"
#include "data\settings.h"

tofrnr, 2018-02-18:
update:
as it turns out, if the fonts folder containing the Monospaced_plain_8.h file was moved from 
libraries\fonts  
into the OLED subdirectory libraries\esp8266-oled-ssd1306\, now appearing as
libraries\esp8266-oled-ssd1306\fonts\Monospaced_plain_8.h
then the include is eventually deteted by 
`#include <fonts/Monospaced_plain_8.h>`

squix78, 2018-03-21:
Does that mean we can close this issue?

closed 2019-12-04 by stale[bot]