ThingPulse/esp8266-oled-ssd1306 #177: The fastest way to make € symbol working?

obykoo, 2018-02-16:
Hello,

 what would be the fastest way to make € symbol working? Only option is to create cutom font?

reivaxy, 2018-02-16:
Hi
You could use my font editor to edit an existing font and replace a character you don't need with €

It's in my fork, hasn't been merged unfortunately:

https://github.com/reivaxy/esp8266-oled-ssd1306/tree/master/resources
Download glyphEditor.html and open it in your browser straight from your drive, no need for a web server

mau-lopez, 2018-11-17:
Hi, first of all thanks for such a great library, really useful and easy to use. This will probably be a very silly question: glyphEditor.html to generate just one character, thermometer glyph. I can generate the output, and have copied it to the bottom of OLEDDisplayFonts.h changing the char type to uint8_t, but I have no idea how to call my only character in Arduino IDE, ¿with which key on my keyboard or with what character code? Thanks a lot in advance!

Arduino IDE 1.8.7
MacBook Pro MacOS 10.14.1

[image]

reivaxy, 2018-11-17:
Hi, 

You need to call the display setFont method and give it the name of the variable in which you defined your don't, looks to be "Termometer" here.
Then call drawString and give it a string made of your character, its code is 1 in the screen shot you made (field "first char code" in the form)
You could also, in the glyphEditor form, give it the code of a displayable character, for instance 65, generate the don't again, and then your icon would be displayed with the string "a" given to drawString.
Hope this makes sense.

mau-lopez, 2018-11-20:
reivaxy, followed your instructions and it worked perfectly. Thank you very much for your time and help!

closed 2019-12-04 by stale[bot]