ThingPulse/esp8266-oled-ssd1306 #179: 128x64 SSD1306 OLED display skips every alternative row when i2c address is changed to 0x3D

chloewyin, 2018-03-08:
Hi, I am conducting a project that needs to use two SSD1306 OLED displays (both 128x64 resolution) with the same SCA SCL pins. I found that it can be achieved by changing the i2c address of one of the OLED display by resoldering the resistor from 0x78 to 0x7A, which will change its address from 0x3C to 0x3D. I did that and tested the resoldered OLED display with i2c scanner script, it detected the i2c device at address 0x3D. 
[image]

However, when i tried to test the resoldered display with SSD1305SimpleDemo.ino, changing the Line 61 from `SSD1306  display(0x3c, D3, D5);` to `SSD1306  display(0x3d, D3, D5);`, the OLED display shows graphics that skip every alternative row as shown in the following photo:
[image]

I tested the SSD1305SimpleDemo.ino with the other original OLED display with address 0x3C, the graphics are perfectly fine. So I'm not sure whether if I have damaged the resoldered OLED display when changing the resistor connection? Or is there any other code that I should change when using i2c address 0x3D? 

Need urgent help please, thanks!

mrmonteith, 2018-03-08:
Not sure this cures the issue or not.  But every solder job I do I follow up with a brush with acetone on it to clean off the rosin from the solder.   You could use a q-tip and alcohol or fingernail polish remover, which is acetone based anyway.   It certainly can't hurt and always a good idea to get into a habit of.

DeuxVis, 2018-03-08:
I would not use acetone near delicate components or anything plastic like - header pins, wires insulation, ribbon wire...

Rather use alcohol - Isopropyl is usually recommended for cleaning PCBs but regular medical alcohol works too.

mrmonteith, 2018-03-08:
Deux, you're right.  I am careful to keep it away from those components.  Thanks

DeuxVis, 2018-03-08:
@chloewyin regarding your trouble, Unluckily I don't have one of those selectable address OLED screen available to try and reproduce it.

This said I suspect you have a hardware problem, somebody would already have it reported around here if that was a software bug.

Do you think it is possible you have overheated something during your soldering operation ? The flexible ribbon that is used to connect the actual screen to the PCB on those oleds are probably not very reliable in their connection.

If I were you I would try to revert back the resistor to its original place to see if the problem maintains. This would mean you have damaged something.

chloewyin, 2018-03-09:
Thanks for the replies!

Actually I have found a way to use multiple OLED displays with the same i2c address, that is by using an i2c multiplexer (TCA9548A). It works perfectly when I connected it to two OLED displays both with address at 0x3C, controlled using ESP8266. 

However, I am still curious with the problem of that resoldered OLED display. I'll try cleaning it with alcohol as soon as I get one. If it still doesn't work, I'll try to resolder the resistor back to 0x78 to see  whether if I've actually damaged it.  Thanks @DeuxVis for the suggestions :) Will get back here when I've gotten any results.

dpharris, 2018-03-09:
I see Adafruit has a breakout for that chip:
https://www.adafruit.com/product/2717

suculent, 2018-03-18:
I have the same issue without even changing the I2C address. Not that it helps...

hellstef, 2018-06-30:
search for the file Adafruit_SSD1306.h on your Computer, I found this file on two different places. so 

#include <Adafruit_SSD1306.h>

in the example program looked at the wrong place where the 128 x 32 Pixel Display was still defined

//   #define SSD1306_128_64
   #define SSD1306_128_32
//   #define SSD1306_96_16

changing the other file to

   #define SSD1306_128_64
//   #define SSD1306_128_32
//   #define SSD1306_96_16

solved the problem

mneuron, 2019-01-10:
The row skipping is a hardware failure. It happens when the lower left or right corner of the glass is broken.
It happend to me twice when inserting the display in a frame.
In the second picture of @chloewyin there is a glass fracture visible in the lower left corner.

Joennuh, 2019-01-10:
@mneuron One of my SSD1306 that I've laying around does also have this issue (without modifications). I will check for damage in de lower corners. Thank you for the hint!

suculent, 2019-01-10:
I can confirm this. Once you have properly setup the display height (make sure you don’t mix 32 and 64 bits/pixels), this can happen for displays that have been broken. Happens to 128x64 with bigger chance as of to device size. 

Odesláno z iPhonu

cbgj, 2019-02-23:
> The row skipping is a hardware failure. It happens when the lower left or right corner of the glass is broken.
> It happend to me twice when inserting the display in a frame.
> In the second picture of @chloewyin there is a glass fracture visible in the lower left corner.

Thanks for this, it's been driving me nuts trying to troubleshoot for hours. Mine also had a small fracture in the corner.

closed 2019-09-05 by stale[bot]

websrvmgr, 2020-12-02:
I had the same issue. It was a problem with the low corners (one or both). Due for shipping y/o manipulation.
From now, what I do with NEW ones (the ones that have problems cannot be fixed), is to put silicone glue in the corners, between the glass and pcb, so I prevent to damage the glass when manipulating the screen. (you can put any other soft thing between glass and pcb that prevents damage when you accidentally push them to pcb)

[image]
[image]
[image]