ThingPulse/esp8266-oled-ssd1306 #181: !Backwards compatiblity break by #97

Rotzbua, 2018-03-14:
This change will fuck up all current implementations, if a new version is released. The first argument `OLEDDISPLAY_GEOMETRY g`  will cause this issue.
Better overload the constructor or set a default value for the argument.
```c++
// old
SSD1306Wire(uint8_t _address, uint8_t _sda, uint8_t _scl) {}

// new changed version
SSD1306Wire(OLEDDISPLAY_GEOMETRY g, uint8_t _address, uint8_t _sda, uint8_t _scl) {}
```

squix78, 2018-03-14:
Hi Rotzbua. Thank you for your feedback. I absolutely agree with you. The plan is to keep old implementations running and overloading the constructor is a very good suggestion for that. I will try to integrate most of the 11 remaining PRs before I will re-establish API compatibility as far as possible. However, the next version will probably be 4.0.0 and by following semver standards indicate major changes in the API. Please feel free to point out again should I have missed something after the release

squix78, 2018-03-15:
I added the constructer overload as you suggested but there is still an API breaking change: all binary resources (fonts, images) have changed from data type **const char** to **const uint8_t**. Any suggestion how we can make this issue smoother? Overloading all functions seems to be a bit brutal but I'm open...

Rotzbua, 2018-03-16:
@squix78 Thanks. I will have a look this weekend.

PR #183 (2018-03-18): improve constructor
fix #181
commit c1e36a11 (2018-03-18): improve constructor

fix #181
closed 2018-03-18 by squix78