ThingPulse/esp8266-oled-ssd1306 #195: invalid conversion from 'const char*' to 'const uint8_t*

dan2600, 2018-03-25:
I'm using the Arduino IDE. I created a new file in my skechbook called fonts.h .  Copy and pasted the results from the webapp into that. In my main file i put at the top "#includes fonts.h" but when i try to call 
  
display.setFont(Monospaced_plain_5);

upon compiling I get the error

invalid conversion from 'const char*' to 'const uint8_t* 

am I doing something wrong?

dan2600, 2018-03-25:
changing char to uint8_t in the header file fixed the problem, the webapp should be updated maybe?

reivaxy, 2018-03-25:
Hello, are you talking about the glyphEditor.html app ?
I did update it and made the PR yesterday :)
https://github.com/ThingPulse/esp8266-oled-ssd1306/pull/193

reivaxy, 2018-03-25:
Ah I guess you were probably talking about the font converter...

spacehuhn, 2018-03-27:
Hey I use the library for a few projects and people reported the same issue, however if I change it I get an error message that uint8 would be wrong.
Was there a library update I haven't noticed or why is this making problems?

Because right now half of the people will complain that my software doesn't compile and if I change it in my program, the other half will complain that it doesn't work. Depending on their setup.

Rotzbua, 2018-03-27:
https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/master/UPGRADE-4.0.md

dan2600, 2018-03-27:
@Rotzbua welp looks like I missed that, thank you.

@reivaxy yes I was referring to the webapp @ http://oleddisplay.squix.ch/#/home 

Last night I started using this Javascript app https://github.com/squix78/esp8266-oled-ssd1306-font-converter, @spacehuhn maybe this will help you too

figured it out from this thread: https://github.com/squix78/esp8266-oled-ssd1306-font-converter/issues/1 

(On windows, requires Java SDK)

1. Download the file from https://github.com/squix78/esp8266-oled-ssd1306-font-converter/blob/master/src/main/java/ch/squix/esp8266/fontconverter/rest/FontConverterV3.java

2. edit FontConverterV3.java remove line 1. 

3. change line 62 to the font you want to convert and size (based on system installed fonts and the names Java sees them as, there are scripts to get a font list from Java if you have trouble, I found that the names Photoshop uses for fonts are the same as Java)

4. Uncomment line 64

5. compile the java file (javac FontConverterV3.java)

6. run the java file and the font will come to STout (java FontConverterV3)

I have a TEE.BAT file on my windows system that allows me to write the stout to a text file, so in the end I ran

java FontConverterV3 | TEE.BAT newfont.txt

hope this helps someone, been working great with pixel fonts!!!!!

closed 2018-03-27 by dan2600