ThingPulse/esp8266-oled-ssd1306 #208: SH1106 flickers while updating but SSD1306 works fine

antonioag95, 2018-07-12:
I'm writing some code which gets temperature and humidity every 2 seconds from DHT22 and then it prints  values on the display. I've been coding using SSD1306 and everything was fine, now I wanted to upgrade my screen to the bigger one and I got that issue. Do you know how to solve it? Thanks!

Here you can see what I'm talking about:
[https://youtu.be/zMpA_k8ndM8](Problem video)

laurentopia, 2018-08-10:
It's a code issue, without seeing it I'd guess that you're calling .display() right after .clear().
The way this lib works is it writes to a frame buffer then dumps it to the display.
So instead, call .display() at the very end of your display code so it would be like this:
.clear() // clears the frame buffer
.doStuff()
.display() // push the frame buffer to the OLED

antonioag95, 2018-08-26:
I got it! It was a code issue but not the one mentioned by @laurentopia . It seems that calling .setContrast(255) causes that annoying bug ONLY when using SH1106, that's why the same code on SSD1306 was fine! I removed those lines and now my display is not flickering anymore. I'm leaving the issue open in case @squix78 wants to explain why this is happening, otherwise just tell me and I'll close it.