ThingPulse/esp8266-oled-ssd1306 #209: Temporarily stop frame animation possible?

millenseer, 2018-08-01:
Let's take the WorldClockDemo as example.
I merged my MQTT-functions in order to display submitted text on the OLED or to blink the LED.
But I am unable to stop the animation, do my stuff and then continue the animation after a certain time.
Doing my display.clear(); display.drawStringMaxWidth(...); display.display() works well, but the frames will interfere and continue automatically after some time.

Am I missing some ui-functions? Should I reserve a frame for my "stuff" and use switchToFrame() ? Will this stop the animation?

Any example dearly welcome.

RudyFiero, 2018-08-02:
Yes you can stop and then restart the frame rotation. I have done this. I added a couple more frames to the rotation. I have a button to pause and restart, but that is only the trigger, and time could be a trigger as well.

It has been a while since I did this and I don't offhand remember how. I'm on laying on the couch with a tablet and it would be too hard to give you an example.

enableAutoTransition();

/**
 * Disable automatic transition to next frame.
 */
void disableAutoTransition();

Ok I looked and I am sure the above is what you want.

millenseer, 2018-08-03:
Ah, thank you for pointing that one out for me. Only now I understand that transition thing. Works great. Thanks!

closed 2018-08-03 by millenseer