ThingPulse/esp8266-oled-ssd1306 #213: Documentation for different resolution oleds

mcer12, 2018-08-10:
I work with the library from time to time and always forgot to add geometry parameter for 128x32 display because it's not documented anywhere. It can be found in the pull request but since 128x32 resolution oleds are common aswell, I think it should be pointed out in readme or demo sketches how to make them work.

ReanimationXP, 2018-08-29:
+1 on this

scropion86, 2018-08-29:
+1 i was just searching for the same i have to dig in the source code to find it also i didn't find any example
showing how to set other geometry.
i have 0.91 Inch from [AE](https://www.aliexpress.com/store/product/0-91-Inch-128x32-IIC-I2C-White-Blue-OLED-LCD-Display-DIY-Module-SSD1306-Driver-IC/1757110_32879702750.html?spm=2114.search0104.3.8.66706e37vgMazb&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_10870_10867_10059_10696_100031_10084_10083_10103_10618_10307,searchweb201603_1,ppcSwitch_5&algo_expid=e03d427d-66ed-4a02-8580-8c3f705cc030-1&algo_pvid=e03d427d-66ed-4a02-8580-8c3f705cc030&priceBeautifyAB=0) which Resolution is 128 x 32 , if somebody have a working sketch kindly share it

mcer12, 2018-08-29:
Instead of

> SSD1306Wire  display(0x3c, D3, D5);

Init with something like this

> SSD1306Wire display(0x3c, D3, D5, GEOMETRY_128_32 ); // WEMOS OLED shield

ReanimationXP, 2018-08-29:
^ this is correct. the only two options are GEOMETRY_128_32 and GEOMETRY_128_64 as of this writing. it defaults to GEOMETRY_128_64.

LennartHennigs, 2018-09-28:
Hi,
could you consider adding different resolutions to the library – or allowing us to define custom resolutions? I am using the [WeMos OLED shield](https://wiki.wemos.cc/products:d1_mini_shields:oled_shield) which offers 64×48 pixels. That'd be great.

Cheers
l.

mcer12, 2018-09-28:
https://github.com/ThingPulse/esp8266-oled-ssd1306/pull/198

LennartHennigs, 2018-09-28:
ah, thx! that helps. 
(will that be merged back into the master branch?)

mcer12, 2018-10-02:
I'm not in possession of such power. Only rep owner can merge pull
requests. But you can make your own fork by all means.

closed 2019-12-04 by stale[bot]