ThingPulse/esp8266-oled-ssd1306 #224: Typo or not documented function in SSD1306SimpleDemo "ipsum\n"

Nobuo1, 2018-10-24:
In SSD1306SimpleDemo 

    display.setFont(ArialMT_Plain_10);
    display.setTextAlignment(TEXT_ALIGN_LEFT);
    display.drawStringMaxWidth(0, 0, 128,
      "Lorem ipsum\n dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore." );

I wonder if "\n" is documented anywhere which other similar functions

tuxBurner, 2018-11-14:
Hmmm why document this. 
This is in general  LF (Line Feed) in most string handling.

Or do you mean that the display can handle those ?

Joennuh, 2018-11-15:
Maybe the documentation could see the functions do support line feed but actually it is a generally known character combination.

closed 2019-12-04 by stale[bot]