ThingPulse/esp8266-oled-ssd1306 #235: I2C conflict

dadone89, 2019-02-25:
Hi, using ESP32 v1.0.1 it seems that this library cause problems in I2C when there are other components. For example i have 1 SSD1306 and an MLX90614. if i use both in the same code nothing work.

```
#include "SSD1306Wire.h"
#include <Adafruit_MLX90614.h>

String TempValString;

SSD1306Wire  display1(0x3c, 21, 22);
Adafruit_MLX90614 mlx = Adafruit_MLX90614();

void setup() {
  Serial.begin(115200);
  display1.init();
  display1.flipScreenVertically();
  display1.setFont(Monospaced_plain_16);
  display1.display();

  mlx.begin();
}

void loop() {

  TempValString = String(mlx.readObjectTempC());
  Serial.println(TempValString);

  display1.drawString(0, 0, TempValString);
  delay(500);
}
```

if i use the Adafruit lib, the problem not appear.
Maybe the new version of ESP32 board software support (v1.0.1) introduced modify at the wire lib, i don't know if this is the cause.

ghcjssla, 2019-06-11:
I get an error for the same reason.
If you have a workaround, share it.

01ivr3, 2019-08-13:
Does anyone know of a solution on how ssd1306 library can be used simultaneously with another device connected to the second i2c port on an ESP32?!

closed 2020-02-23 by stale[bot]

tochinet, 2020-03-12:
Well, sorry not to solve your problem, but I sadly think that this kind of situation will only improve when the diverging libraries under control of so many programmers will be seriously refactored. The big trouble is that nobody is wanting to spend the time needed for that, because they won't make any money from it.

To illustrate my point: this repo is "esp8266-oled-ssd1306". You're using esp32. So you should be surprised that it works at all, since testing should only be expected on esp8266.

Haustuer, 2025-01-31:
I could solve a similar issue by reducing the I2C Speed

`
 Wire.setClock(100000);   // Speed of I2C bus 100KHz`