ThingPulse/esp8266-oled-ssd1306 #237: Glyph Editor Additional Formats?

dagnall53, 2019-03-08:
Hi, Thanks for a great set of tools. I am trying to update some old software the used a different OLED driver, and the Fonts are VERY similar to Squix format, but I just cannot find any way to convert them. 
They have  bitmap indexes rather than jump tables, and I have run out of talent trying to convert them!. 
Can you add a converter to Glyph editor, or point me to a program that can convert them? They were generated originally by "Font and Bitmap Generator" but this is no longer available.
I have put part of the file structure below in the hope you will recognize it or be able to add it to the Editor. 
MANY MANY thanks
Dagnall

The files start like this:
prog_uchar FONT_5x5iz[] = {
	// Font Header 
	0x01,	// 0x00: Monospace, 0x01: Proportional 
	0x02,	// 0x00: horizontalLeft,  0x01: horizontalRight 
			// 0x02: verticalCeiling, 0x03: verticalBottom 
	0x20,	// Startcharacter 
	0xE0,	// Number of Bitmaps 
	0x8,	// Height 

	// Bitmap Index: Number of Index Entries 224
 	0x1, 0xC5, //Index Nummer: 0
	0x1, 0xC7, //Index Nummer: 1
	0x1, 0xC9, //Index Nummer: 2
	0x1, 0xCD, //Index Nummer: 3
	0x1, 0xD3, //Index Nummer: 4
	0x1, 0xD8, //Index Nummer: 5
	0x1, 0xDE, //Index Nummer: 6
 then have 

	/* Character Data (width: 1  bit) - Index: 0 */
	0x1, 0x7F,
	/* Character Data (width: 1  bit) - Index: 1 */
	0x1, 0x51,
	/* Character Data (width: 3  bit) - Index: 2 */
	0x3, 0x79,0x7F,0x79,

reivaxy, 2019-03-08:
Hi @dagnall53 
Could you provide the bitmaps of the characters in your example?
Also, the whole file would be better than a short extract, if you can make it downloadable from somewhere (or is this the whole file ?)
Thanks

dagnall53, 2019-03-08:
The Bitmaps as I understand are the "character data", first byte is width.. rest is the bitmap for that character. I Only put a quick selection in the previous post as the file is LONG.
I keep a copy of my version of the fonts on github, so I have put the "original" version in this strange format there as well as:  https://github.com/dagnall53/Fonts-for-WiRocs/blob/master/rocfontsall.h  
This has all the fonts (and some bitmaps) I am trying to adapt.   
Many thanks
Dagnall

lorol, 2019-03-08:
See Here:
https://forum.arduino.cc/index.php?topic=447956
https://github.com/lorol/esp8266-oled-ssd1306#changes-by-lorol

dagnall53, 2019-03-08:
I have tried both those with no luck (or maybe no skill?) 
Also the https://rawgit.com link will be lost soon, so I hope Squix will host that nice glyph editor!

reivaxy, 2019-03-08:
Actually by "bitmap" I meant the graphical representation of the characters ;)

dagnall53, 2019-03-08:
I will try and get a few images tomorrow. 
Thanks for looking at this

dagnall53, 2019-03-09:
@reivaxy @lorol I have put images of the start of the character sets as shown by the original font and bitmap editor, as JPG in https://github.com/dagnall53/Fonts-for-WiRocs.. At least you can see there what they should look like. Hopefully this will enable you to "disentangle" the font formatting enough to change them to Squix fonts.
Good luck and thanks for looking!

closed 2019-03-16 by dagnall53