ThingPulse/esp8266-oled-ssd1306 #239: Online Glyph Editor link will be closing

dagnall53, 2019-03-13:
**Started a separate Issue to make sure this gets noticed:** 
The Online GlyphEditor ( https://rawgit.com/lorol/esp8266-oled-ssd1306/master/resources/glyphEditor.html) seems slightly more recent than the one in resources: 
In particular it has the ability to show and modify character widths, which is very useful. 

Because  https://rawgit.com link will be closing soon: can you please update the version in repository?  

> GitHub repositories that served content through RawGit within the last month will continue to be served until at least October of 2019. URLs for other repositories are no longer being served.
> 
> If you're currently using RawGit, please stop using it as soon as you can.

lorol, 2019-03-15:
Please you are welcome to copy, clone modify - the file is simple HTML + JS and can be transferred anywhere for easy web access by community. It should work on your local copy as well (Mozilla)

dagnall53, 2019-03-23:
Many thanks, i popped a copy on my github site so its still available if the original link dies.. 
Ps, the one i used has the proportional spacing editing, which is very useful. The one on the ssd1306 github seems older? 
Dagnal

closed 2019-10-03 by stale[bot]