ThingPulse/esp8266-oled-ssd1306 #245: panic on ESP8266 2.5.0

roncoa, 2019-04-28:
Sometimes Arduino ide gives me the following error.
With Arduino IDE v1.8.9 and ESP8266 2.5.0 and 2.4.2

Panic C:\Users\ronco\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp:103 __yield

ctx: sys 
sp: 3fffe970 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffeb20:  00000095 3fff3219 00000000 00000030  
3fffeb30:  4020278f 00000000 0000002f 0000007f  
3fffeb40:  0000007e 6000001c ff000000 3fff02f8  
3fffeb50:  00000010 00000010 00000000 40100575  
3fffeb60:  0000004e 3fff30a2 00000000 3fff02f8  
3fffeb70:  33323130 37363534 62613938 3fff02f8  
3fffeb80:  00000010 00000010 00000000 40212ac1  
3fffeb90:  00000010 00000010 00000000 4020bb99  
3fffeba0:  00000000 00000003 00000020 00000000  
3fffebb0:  3fff286c 00000384 00000489 00000000  
3fffebc0:  00000004 00000000 00000000 00000009  
3fffebd0:  00000000 00000000 00000011 00000000  
3fffebe0:  00000000 00000020 00000031 40211d9c  
3fffebf0:  00000010 3fffec50 3ffefc34 00000000  
3fffec00:  00000013 3fff286c 3ffefc34 4020be90  
3fffec10:  00000000 00000000 00000000 40211e5d  
3fffec20:  00000000 00000004 3fff286c 40211f60  
3fffec30:  00000003 00000001 3ffefc34 3fffecc0  
3fffec40:  00000003 00000001 3ffefc34 40203a42  
3fffec50:  3fff0c34 0000000f 00000004 40211db8  
3fffec60:  3fff02f8 0000008b 00000000 40204a44  
3fffec70:  00000000 00000000 00000000 401006dc  
3fffec80:  00000003 3fffece4 3ffefbf8 40211d9c  
3fffec90:  00000003 3fff244c 0000000f 00000000

marcelstoer, 2019-06-05:
The issue title mentions 2.5 but this

> C:\Users\ronco\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2

clearly shows you are using 2.4.2.

roncoa, 2019-06-05:
I've tried several cores, but the problem is the same

CyanoFresh, 2019-07-02:
same. I think it's related to the async staff

marcelstoer, 2019-07-03:
Did you ever try to [decode the stacktrace](https://github.com/me-no-dev/EspExceptionDecoder) to see where this points to?

CyanoFresh, 2019-07-03:
```
Decoding stack results
0x40206b6d: __yield() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 105
0x402058d5: OLEDDisplay::drawStringInternal(short, short, char*, unsigned short, unsigned short) at C:\Users\Alex\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src\OLEDDisplay.cpp line 928
0x4020c883: strtok at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/string/strtok.c line 102
0x40205c2f: OLEDDisplay::drawString(short, short, String) at C:\Users\Alex\Documents\Arduino\libraries\esp8266-oled-ssd1306-master\src\OLEDDisplay.cpp line 512
0x402066b0: String::copy(char const*, unsigned int) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 211
0x402066f0: String::String(char const*) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 36
0x402012a0: onMqttConnect(bool) at C:\Users\Alex\Documents\Arduino\FullyFeatured-ESP8266/FullyFeatured-ESP8266.ino line 58
0x402084d4: _umm_free(void*) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1304
0x402088fc: free(void*) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x40210d9a: operator delete(void*) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/del_op.cc line 48
0x40208c11: std::_Function_handler ::_M_invoke(std::_Any_data const&, bool) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40203408: AsyncMqttClient::_onConnAck(bool, unsigned char) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2174
0x402011c8: onMqttConnect(bool) at C:\Users\Alex\Documents\Arduino\FullyFeatured-ESP8266/FullyFeatured-ESP8266.ino line 42
0x40208c18: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 1934
0x40208c04: std::_Function_handler ::_M_invoke(std::_Any_data const&, bool) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40208e91: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, bool, unsigned char) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x402042f0: AsyncMqttClientInternals::ConnAckPacket::parseVariableHeader(char*, unsigned int, unsigned int*) at C:\Users\Alex\Documents\Arduino\libraries\async-mqtt-client-master\src\AsyncMqttClient\Packets\ConnAckPacket.cpp line 25
0x4020336d: AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int) at C:\Users\Alex\Documents\Arduino\libraries\async-mqtt-client-master\src\AsyncMqttClient.cpp line 437
0x4020241c: std::_Function_base::_Base_manager    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40208e6c: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, bool, unsigned char) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40100ba4: millis() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 186
0x40215a29: tcp_output_alloc_header_common at core/tcp_out.c line 1825
0x402033b4: std::_Function_handler ::_M_invoke(const std::_Any_data &, void *, AsyncClient *, void *, unsigned int) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40205054: AsyncClient::_recv(tcp_pcb*, pbuf*, long) at C:\Users\Alex\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp line 420
0x402088fc: free(void*) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x40205098: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at C:\Users\Alex\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp line 498
0x40214828: tcp_input at core/tcp_in.c line 501
0x40100e60: pvPortMalloc(size_t, char const*, int) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\heap.cpp line 155
0x40208718: malloc(size_t) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1677
0x40210000: _fclose_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/fclose.c line 108
0x40219515: ip4_input at core/ipv4/ip4.c line 710
[... cut ...]
0x40208c80: std::_Function_handler ::_M_invoke(std::_Any_data const&, unsigned short) at c:\users\alex\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40206a9d: esp_schedule() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x40206b32: loop_wrapper() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 127
```

used latest async mqtt https://github.com/marvinroger/async-mqtt-client and latest esp8266-oled-ssd1306

closed 2020-01-13 by stale[bot]