ThingPulse/esp8266-oled-ssd1306 #246: Setting geometry (128x32) - how?

Ing-Med, 2019-04-28:
Forgive my stupidity, but i was not able to set the geometry correctly without tinkering in the driver itself. Could you please explain to me how i would do that?

I tried changing all relevant variables in the OLEDDisplay.h but that didn't work for me. Setting OLEDDISPLAY_GEOMETRY geometry              = GEOMETRY_128_64; to GEOMETRY_128_32 didn't work either. 
I only got it working when i added the private setGeomentry function to the init() function. 
setGeometry(GEOMETRY_128_32);

But, well, i shouldn't be tinkering around in the drivers, eh? :-D 

Thank you in advance.

holgi-s, 2019-05-12:
It can be done in the constructor with: 
SSD1306Wire display(0x3c, D3, D5, GEOMETRY_128_32 );

There is already an open Pull Request about updating the documentation regarding this issue.
https://github.com/ThingPulse/esp8266-oled-ssd1306/issues/213#issuecomment-416952257

Ing-Med, 2019-05-13:
Thank you for your reply. I figured it out eventually. I only looked in the oled library but never in the display.... Rookie mistake

closed 2019-05-13 by Ing-Med