ThingPulse/esp8266-oled-ssd1306 #272: Function to set I2C-clock?

Elmi77, 2020-02-08:
Hi,
is there a function to specify the I2C clock to be used?
Explanation: I have a somewhat long cable from ESP32 to the display which causes troubles with a too high I2C frequency, so it would be good to run it at slower speeds.
Thanks!

codebeat-nl, 2020-02-10:
It is not the frequency, it is the capacitance because the cable is too long and probably not shielded and because it is not shielded it acts like an antenna and receives outside noise.  Also possible is that the resistance of the cable reduces the HIGH state of the signal, a voltage drop because of the length and quality of the cable, it is not high enough to determine the high and low state, it starts to malfunction.
To change the clockspeed, take a look at I2C object itself. If you are using arduino, take look here:
https://www.arduino.cc/en/Reference/WireSetClock

Remember that this effects ALL of the connected I2C pheripherials.

Elmi77, 2020-02-10:
The cable is shielded. And yes, it is the capacity AND the inductivity of the cable - which have an as stronger effect as higher the frequency is, thus transmitting data with a lower frequency helps.

Thanks for the hint with the setClock()-function, that solved my issue!

closed 2020-02-21 by marcelstoer