ThingPulse/esp8266-oled-ssd1306 #278: Circles look weird

NoobTracker, 2020-04-19:
Hello!

I have a problem with fillCircle(). It looks like the circles are a pixel higher then wide. I don't have problems with drawCircle().
My code:
```
#include <Wire.h>
#include "SH1106Wire.h" 

SH1106Wire display(0x3c, SDA, SCL);

void setup() {
  display.init();
  display.flipScreenVertically();
}
void loop(){
  display.clear();
  display.setColor(WHITE);
  display.drawCircle(60, 28, 4);
  display.drawCircle(70, 28, 3);
  display.fillCircle(60, 38, 4);
  display.fillCircle(70, 38, 3);
  display.display();
}
```
You can see the result in the attached image.

[image]

marcelstoer, 2020-04-19:
Maybe `fillCircle()` is plagues by the same issues @DaveSprague fixed in #197 for `drawCircle()`.

NoobTracker, 2020-04-19:
No, I had another problem. And I fixed it. I changed

```
void OLEDDisplay::fillCircle(int16_t x0, int16_t y0, int16_t radius) {
  int16_t x = 0, y = radius;
	int16_t dp = 1 - radius;
	do {
		if (dp < 0)
      dp = dp + (x++) * 2 + 3;
    else
      dp = dp + (x++) * 2 - (y--) * 2 + 5;

    drawHorizontalLine(x0 - x, y0 - y, 2*x);
    drawHorizontalLine(x0 - x, y0 + y, 2*x);
    drawHorizontalLine(x0 - y, y0 - x, 2*y);
    drawHorizontalLine(x0 - y, y0 + x, 2*y);

	} while (x < y);
  drawHorizontalLine(x0 - radius, y0, 2 * radius);

}
```
to
```
void OLEDDisplay::fillCircle(int16_t x0, int16_t y0, int16_t radius) {
  int16_t x = 0, y = radius;
	int16_t dp = 1 - radius;
	do {
		if (dp < 0)
      dp = dp + (x++) * 2 + 3;
    else
      dp = dp + (x++) * 2 - (y--) * 2 + 5;

    drawHorizontalLine(x0 - x, y0 - y, 2*x + 1);
    drawHorizontalLine(x0 - x, y0 + y, 2*x + 1);
    drawHorizontalLine(x0 - y, y0 - x, 2*y + 1);
    drawHorizontalLine(x0 - y, y0 + x, 2*y + 1);

	} while (x < y);
  drawHorizontalLine(x0 - radius, y0, (2 * radius) + 1);  
  drawVerticalLine(x0, y0 - radius, 2 * radius);
}
```
I'm not sure, but I think that the problem was caused by horizontalLine(). I guess this should be implemented in a future bugfix. Now it looks much better.

[image]

NoobTracker, 2020-04-19:
Hopefully this will be fixed in an update soon, because I want to use this lib in a game and don't want to tell the people "Please delete line x to y and insert this:".

marcelstoer, 2020-04-20:
By far the quickest way to get issues resolved is through pull requests.

NoobTracker, 2020-04-22:
I haven't figured out how these pull requests work. I'm new to GitHub and my english is not the best.

marcelstoer, 2020-04-22:
Fear not - there's a first time for everything, we've all been there 😄 Finding your way around GitHub and knowing how to create PRs is a powerful tool in your developer toolbox. And very satisfactory, too!

Can I encourage you maybe start with this video https://youtu.be/For9VtrQx58 or our own [guidelines](https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/master/CONTRIBUTING.md)?

NoobTracker, 2020-04-23:
That's really interesting! Thank you!

closed 2020-11-07 by stale[bot]