ThingPulse/esp8266-oled-ssd1306 #281: Tiny fonts

NoobTracker, 2020-04-24:
I have a problem creating new fonts. All fonts under size 8 are really unreadable. And size means width. This means that each letter is 10 pixels high. This is annoying, because I switched from the Adafruit GFX to this lib primarily because of the font functions. And the preview on the font website is irritating because the font is not pixelated as it actually looks. However, I would need a font that is 8 pixels high so that I can reprogram the lib so that only individual areas are updated and these are 8 pixels high. I would very much like to have a few additional standard fonts, such as 5x7 (6x8 with spacing), 5x7 with variable width and 3x5 (4x6) with variable width. I would also like to see the preview pixelated down to 128x64. Would it be possible?

unrzn0, 2020-09-20:
I had a similar issue, as I wanted to have some tick labels. Finally I found this here: https://blog.mclemon.io/hacking-a-tiny-new-font-for-the-ssd1306-128x64-oled-screen , which leads to the Tom Thumb font from https://robey.lag.net/2010/01/23/tiny-monospace-font.html . Unfortunately the format is BDF, which I've managed to convert to the format used by this library (using an Excel-based tool from https://forum.arduino.cc/index.php?topic=447956 ) as results in the attached snippet. Simply copy this into the OLEDDisplayFonts.h file (or include your own file with that content). Note that according to the authors Brian Swetland and Robey Pointer this font is licensed according to the CC0 or CC-BY 3.0 license.

`
const uint8_t TomThumb4x6[] PROGMEM = {
0x03, // Width: 3
0x06, // Height: 6
0x20, // First char: 32
0xBD, // Number of chars: 189
// Jump Table:
0xFF,0xFF,0x00,0x02, // space
0x00,0x00,0x02,0x03, // exclam
0x00,0x02,0x03,0x03, // quotedbl
0x00,0x05,0x03,0x03, // numbersign
0x00,0x08,0x03,0x03, // dollar
0x00,0x0B,0x03,0x03, // percent
0x00,0x0E,0x03,0x03, // ampersand
0x00,0x11,0x02,0x03, // quotesingle
0x00,0x13,0x03,0x03, // parenleft
0x00,0x16,0x02,0x03, // parenright
0x00,0x18,0x03,0x03, // asterisk
0x00,0x1B,0x03,0x03, // plus
0x00,0x1E,0x02,0x03, // comma
0x00,0x20,0x03,0x03, // hyphen
0x00,0x23,0x02,0x03, // period
0x00,0x25,0x03,0x03, // slash
0x00,0x28,0x03,0x03, // zero
0x00,0x2B,0x02,0x03, // one
0x00,0x2D,0x03,0x03, // two
0x00,0x30,0x03,0x03, // three
0x00,0x33,0x03,0x03, // four
0x00,0x36,0x03,0x03, // five
0x00,0x39,0x03,0x03, // six
0x00,0x3C,0x03,0x03, // seven
0x00,0x3F,0x03,0x03, // eight
0x00,0x42,0x03,0x03, // nine
0x00,0x45,0x02,0x03, // colon
0x00,0x47,0x02,0x03, // semicolon
0x00,0x49,0x03,0x03, // less
0x00,0x4C,0x03,0x03, // equal
0x00,0x4F,0x03,0x03, // greater
0x00,0x52,0x03,0x03, // question
0x00,0x55,0x03,0x03, // at
0x00,0x58,0x03,0x03, // A
0x00,0x5B,0x03,0x03, // B
0x00,0x5E,0x03,0x03, // C
0x00,0x61,0x03,0x03, // D
0x00,0x64,0x03,0x03, // E
0x00,0x67,0x03,0x03, // F
0x00,0x6A,0x03,0x03, // G
0x00,0x6D,0x03,0x03, // H
0x00,0x70,0x03,0x03, // I
0x00,0x73,0x03,0x03, // J
0x00,0x76,0x03,0x03, // K
0x00,0x79,0x03,0x03, // L
0x00,0x7C,0x03,0x03, // M
0x00,0x7F,0x03,0x03, // N
0x00,0x82,0x03,0x03, // O
0x00,0x85,0x03,0x03, // P
0x00,0x88,0x03,0x03, // Q
0x00,0x8B,0x03,0x03, // R
0x00,0x8E,0x03,0x03, // S
0x00,0x91,0x03,0x03, // T
0x00,0x94,0x03,0x03, // U
0x00,0x97,0x03,0x03, // V
0x00,0x9A,0x03,0x03, // W
0x00,0x9D,0x03,0x03, // X
0x00,0xA0,0x03,0x03, // Y
0x00,0xA3,0x03,0x03, // Z
0x00,0xA6,0x03,0x03, // bracketleft
0x00,0xA9,0x03,0x03, // backslash
[... 294 lines cut ...]
0x08,0x1D,0x1E, // egrave
0x0A,0x1D,0x1C, // eacute
0x09,0x1D,0x1D, // ecircumflex
0x09,0x1C,0x1D, // edieresis
0x00,0x1D,0x02, // igrave
0x02,0x1D, // iacute
0x01,0x1D,0x01, // icircumflex
0x01,0x1C,0x01, // idieresis
0x0A,0x17,0x1D, // eth
0x1D,0x07,0x1A, // ntilde
0x08,0x15,0x0A, // ograve
0x0A,0x15,0x08, // oacute
0x09,0x15,0x09, // ocircumflex
0x09,0x17,0x0A, // otilde
0x09,0x14,0x09, // odieresis
0x04,0x15,0x04, // divide
0x1C,0x16,0x0E, // oslash
0x0D,0x12,0x1C, // ugrave
0x0C,0x12,0x1D, // uacute
0x0D,0x11,0x1D, // ucircumflex
0x0D,0x10,0x1D, // udieresis
0x04,0x2A,0x1D, // yacute
0x3E,0x14,0x08 // thorn
};
`

marcelstoer, 2021-04-25:
I have been thinking about how to make your new font available to users of this project. As this is a library continuing to package out-of-the-box fonts into `OLEDDisplayFonts.h` does not scale well. Some options:

- One `.h` file per font (including yours). Developers will have to include every font the want to use.
- Add a "Fonts" chapter to the README with links to fonts not available out of the box (e.g. yours).

What do you think?

unrzn0, 2021-04-26:
I would recommend to add documentation how to convert fonts and in addition bundle some thare license-wise okay as one .h file per font. This would make extending easy but also make clear what fonts arre available. Not sure memory is an issue with many fonts, I guess compilers strip what‘s not used.

marcelstoer, 2021-04-27:
> add documentation how to convert fonts

Can you maybe add a paragraph or two here as a comment so I know what _you_ (as a user) would like to read? I can take it from there. I guess the documentation would have to include links to http://sourpuss.net/projects/fontedit/ and https://oleddisplay.squix.ch/.

unrzn0, 2021-04-27:
Hm, not sure I can help so much with this: I’d prefer if you could document the format of the fonts (seems to be different from other formats - or is it the same as an already documented one?) and then my old notes hint at a tool how to convert from the standard BDF format to your format.

closed 2022-04-30 by stale[bot]