ThingPulse/esp8266-oled-ssd1306 #283: platformio ESP32 loops in boot.

loulou4cv, 2020-05-01:
Hello,

I have installed library in ArduinoIDE : no problem.
I have installed library in VSCode/Platformio : no compilation problem but any access to a display function makes the ESP32 booting permanently.

platformio : up to date, core 4.3.3 with Espressif32 platform 1.12.1
library : 4.1.0

I declared
#include <Arduino.h>
#include <stdint.h>
#include "WiFi.h"
#include <SPI.h>
#include <Wire.h>
#include <SSD1306Wire.h>

SSD1306Wire display(0X3C, 5, 4);

whatever function is loop-booting the ESP32, even having just display.clear(); in the setup function.

Any idea please ?

loulou4cv, 2020-05-01:
Hello,

I have found the problem : I forgot to call display.init(); at setup. Then ESP32 was booting all the time.
You can close it. Thank you.

marcelstoer, 2020-05-01:
> You can close it.

Glad you found the bug yourself.

closed 2020-05-01 by marcelstoer