ThingPulse/esp8266-oled-ssd1306 #287: How about adding setTextColor() Function?

yammayamm, 2020-05-31:
Why don't you have setTextColor(FontColor,BackgroundColor) function? I was trying to use this function to display the inverted text for my project, but I couldn't use it. It seems that you don't have even basic function of changing text color neither. Can you add this function? Thx :)

sjly3k, 2020-05-31:
Cool :) I found this function in this package but I couldn't. If this function works in this package, All users who use this package can make more fancier haha XD

marcelstoer, 2020-05-31:
> Why don't you have setTextColor(FontColor,BackgroundColor) function?

We don't need it and nobody else requested it so far. However, we would certainly welcome PR to that effect by you or somebody else.

SVLoneStar, 2020-11-10:
Hi - sure, this is not desperately needed. But it would be nice to be able to show lines of 'inverted' text on the display.
Any other means to do this? If not, implementing this feature would be highly appreciated (by me). Thanks!

turbofex, 2020-12-24:
vote yes on set text color,very useful when making menus.

turbofex, 2020-12-24:
found a way around this.draw the strings,then call setColor(INVERSE) and fillRect() over the text area would do a inverted text effect.works for me.
U could give it a try. @yammayamm

yammayamm, 2020-12-28:
Thnks :)

closed 2020-12-28 by yammayamm