ThingPulse/esp8266-oled-ssd1306 #294: Menu Change between ui.setFrames

beocycris, 2020-07-16:
Hello from Germany.
Right now I´m trying to implement a ui (ESP32 and SSD1306), where the user can change between different menu states and menu levels. To do that, I wrote functions that change to another FrameCallback array, via ui.setframes().
Just like:

FrameCallback frames_controles[] ={fOneControle,fTwoControle,fThreeControle,fBack };
via
ui.setFrames(frames_controles, frameCount_controles);
 
to the next level:

FrameCallback frames[] = {drawFrame1, drawFrame2, drawFrame3, drawFrame4, drawFrame5};
via
ui.setFrames(frames, frameCount);

Now the issue.
After a while, when I press my rotary encoder to change the frame level, the display stocks for about 33 seconds. This problem is not reproducible. A few times it occurs directly - a few times after 5-10 changes.

Does anyone have the same issues and can help me?

Thanks in advance.

regards
Frank

CodeMittens, 2020-10-09:
Hi Frank, did you ever find a way around this?  I see very similar behaviour...

neutralvibes, 2020-10-14:
I have the same issue. 

Switching from a home screen transitioning between different data display frames, then displaying menu lists based on a single frames. It can be instant, then on occasion the library decides to take a number of seconds to actually call the 'draw' function for the new frame(s) specified in the 'UI.setFrames()' func. I verified that by sending a debug message when the frame is first called.

A quick look at the code shows that 'UI.setFrames()' also calls a 'reset' method that may not take into account all the variables that need resetting, but I have had no luck in working out what is missing as yet.

Would be great if someone who has a good understanding of the mechanics could take a look as it may be a quick fix.

closed 2021-06-18 by stale[bot]

thijstriemstra, 2021-07-24:
Is this related to pull request #273?

beocycris, 2021-07-25:
Oh perfect. You’re the man. Thank you.