ThingPulse/esp8266-oled-ssd1306 #295: Could not compile in PlatformIO on ESP32

robig, 2020-07-16:
The example is running fine in Arduino IDE.
But when trying to work with PlatformIO I'm getting a lot of compiler errors:

`In file included from src/Display.cpp:20:0:
.pio/libdeps/lolin32/ESP8266_SSD1306/src/SH1106Wire.h:51:24: error: expected ')' before '_address'
     SH1106Wire(uint8_t _address, uint8_t _sda, uint8_t _scl, OLEDDISPLAY_GEOMETRY g = GEOMETRY_128_64) {
                        ^
.pio/libdeps/lolin32/ESP8266_SSD1306/src/SH1106Wire.h:43:18: error: an anonymous struct cannot have function members
 class SH1106Wire : public OLEDDisplay {
                  ^
.pio/libdeps/lolin32/ESP8266_SSD1306/src/SH1106Wire.h:43:18: error: member 'OLEDDisplay <anonymous class>::<anonymous>' with constructor not allowed in anonymous aggregate
.pio/libdeps/lolin32/ESP8266_SSD1306/src/SH1106Wire.h:43:18: error: member 'OLEDDisplay <anonymous class>::<anonymous>' with destructor not allowed in anonymous aggregate
.pio/libdeps/lolin32/ESP8266_SSD1306/src/SH1106Wire.h:43:18: error: member 'OLEDDisplay <anonymous class>::<anonymous>' with copy assignment operator not allowed in anonymous aggregate
.pio/libdeps/lolin32/ESP8266_SSD1306/src/SH1106Wire.h:173:2: error: abstract declarator '<anonymous class>' used as declaration
 };`

Bert-code, 2021-01-03:
> In file included from src/Display.cpp:20:0: .pio/libdeps/lolin32/ESP8266_SSD1306/src/SH1106Wire.h:51:24: error: expected ')' before '_address' SH1106Wire(uint8_t _address, uint8_t _sda, uint8_t _scl, OLEDDISPLAY_GEOMETRY g = GEOMETRY_128_64)
Are you sure you are not missing the ) somewhere?

marcelstoer, 2021-01-03:
https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/master/src/SH1106Wire.h#L68

Those are parameter default values. The CI build is green: https://travis-ci.org/github/ThingPulse/esp8266-oled-ssd1306

closed 2021-07-23 by stale[bot]