ThingPulse/esp8266-oled-ssd1306 #296: Not working for esp8266 with software i2c

zanjie1999, 2020-07-19:
2020-01 version is working

The latest version is not working

SSD1306Wire  display(0x3c, D4, D5);

D4=gpio2

D5=gpio14

meltdown03, 2020-08-05:
```
In file included from /home/neal/Arduino/google-iot/google-iot.ino:4:0:
/home/neal/Arduino/libraries/ESP8266_SSD1306/src/SSD1306Wire.h: In constructor 'SSD1306Wire::SSD1306Wire(uint8_t, int, int, OLEDDISPLAY_GEOMETRY, HW_I2C, int)':
/home/neal/Arduino/libraries/ESP8266_SSD1306/src/SSD1306Wire.h:76:47: error: 'Wire1' was not declared in this scope
       this->_wire = (_i2cBus==I2C_ONE)?&Wire:&Wire1;
                                               ^
```

This is the error I get when trying to build the new release for my ESP8266. I think it needs another `#if ESP8266` preprocessor since the esp8266 only has the one `TwoWire` class by default.

marcelstoer, 2020-08-06:
> ...the new release...

You mean after the recent changes by @benoitm974?

meltdown03, 2020-08-06:
> > ...the new release...
> 
> You mean after the recent changes by @benoitm974?

Yes, I believe this line is the issue:
https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/65615be7dbd2dc8f1a0477ca1fe93c296d20ef99/src/SSD1306Wire.h#L76

benoitm974, 2020-08-06:
The original issue is 18days old, when my merge was not there yet, si I believe this issue is a different ticket

BUT still my code seems not proof for non ESP32 code, I've submitted a pull request, but I don't have ESP8266 to test.

meltdown03, 2020-08-06:
Yes, sorry, I don't think it's the same issue. The fix (#298) does work for my issue though, so thank you!

marcelstoer, 2020-08-07:
> The original issue is 18days old, when my merge was not there yet, so I believe this issue is a different ticket

I was also surprised and hence I asked for Neal's confirmation. However, his compiler clearly showed there something wrong. Thanks for the fix.

jkunkee, 2020-09-03:
Another datapoint that might be useful here:

I have an ESP8266 project with an SSD1306 attached via I2C alongside a device that can only operate at 100kHz. The latest release (4.1.0 in the Arduino library manager, without this fix) always fails to communicate with one device or the other depending on whether I reset the clock speed after SSD1306 initialization using `Wire.setClock`.

I pulled down the main branch with #298 merged and reworked my code to call the new constructor to set the clock speed to 100kHz. Now, a) I no longer see the individual I2C transmit and receive failures I was seeing before and b) both devices work reliably.

(Seriously, @benoitm974, I can't thank you enough.)

benoitm974, 2020-09-03:
You're welcome I added this constructor for exactly this issue of having my
temp sensor limited to 100khz!

Have fun.

closed 2021-04-25 by marcelstoer