ThingPulse/esp8266-oled-ssd1306 #313: Not working with Ticker library (timer)

pabloquimbar, 2020-11-20:
Hi,
I programmed a code where some sensors are read by I2C and then the string generated with the sensor's output is sent to the OLED. It works perfectly and I thank you for your amazing library.
However, I added now an interrupt with the Ticker library and ask the nodeMCU to do the same. However, this time there are no delays in the code. Just an ISR.
Everything seems to work except for the OLED. It gives a problem that resets the microcontroller.
If I get rid of the function that calls the OLED writing instructions and put it back in the loop, it works!! 

Do you know what may be the problem here? Seems there is some incompatibility. I added an screenshot of the serial port. As you can see, on top there are the measurements of the sensors (in spanish) and then, when I have to write the value in the OLED, the program crashes.

[image]

pabloquimbar, 2020-11-20:
I could add the code if you want. It's in spanish and splitted in different .ino, but I think is understandable.

closed 2021-06-18 by stale[bot]

isaac-bang, 2021-06-19:
hello