ThingPulse/esp8266-oled-ssd1306 #317: setTimePerTransition(0) causes unexpected one pixel frame shift - please apply my bugfix

AlexeyMal, 2021-01-03:
**in OLEDDisplayUi.cpp
line 304
please apply this correction (to avoid division by zero):**

void OLEDDisplayUi::drawFrame(){
  switch (this->state.frameState){
     case IN_TRANSITION: {
		**float progress = 0.f; //AMA fix
		if (this->ticksPerTransition > 0u) progress = (float) this->state.ticksSinceLastStateSwitch / (float) this->ticksPerTransition; //AMA fix**
       //float progress = (float) this->state.ticksSinceLastStateSwitch / (float) this->ticksPerTransition;

marcelstoer, 2021-01-03:
In what case will `ticksPerTransition` ever be null?

AlexeyMal, 2021-01-04:
ticksPerTransition is null when I set setTimePerTransition(0); to get immediate change of frames. It works well after my fix. Thanks!-------- Исходное сообщение --------Тема: Re: [ThingPulse/esp8266-oled-ssd1306] setTimePerTransition(0) causes unexpected one pixel frame shift - please apply my bugfix (#317)От: Marcel Stör Кому: ThingPulse/esp8266-oled-ssd1306 Копия: AlexeyMal ,Author 
In what case will ticksPerTransition ever be null?

commit b730115a (2021-01-04): Avoid division by zero

Fixes #317
closed 2021-01-04 by marcelstoer