ThingPulse/esp8266-oled-ssd1306 #321: late ui init show blank screen until 65sec.

hyeok, 2021-01-14:
below is SSD1306UiDemo. I just added delay for 40sec before ui init.

`... ... ...`
`void setup() {`
`  Serial.begin(74880);`
`  Serial.println();`
`  Serial.println();`
`  **Serial.println("delay for 40s"); // added for test**`
`  **delay(40000); // added for test**`
`  **Serial.println("delay Finished"); // added for test**`
`  ui.setTargetFPS(60);`
`  ui.setActiveSymbol(activeSymbol);`
`  ui.setInactiveSymbol(inactiveSymbol);`
`  ui.setIndicatorPosition(BOTTOM);`
`  ui.setIndicatorDirection(LEFT_RIGHT);`
`  ui.setFrameAnimation(SLIDE_LEFT);`
`  ui.setFrames(frames, frameCount);`
`  ui.setOverlays(overlays, overlaysCount);`
`  ui.init();`
`  display.flipScreenVertically();`
`}`
`void loop() {`
`  int remainingTimeBudget = ui.update();`
`  if (remainingTimeBudget > 0) {`
`    delay(remainingTimeBudget);`
`  }`
`}`

UI doesn't show till 65sec from boot.

In OLEDDisplayUi.cpp...

`int16_t timeBudget = this->updateInterval - (frameStart - this->state.lastUpdate);`

if ui inited after 32sec, timeBudget is plus, so doen't draw ui, util 65sec.

I think timeBudget should be int32_t.

`int32_t timeBudget = this->updateInterval - (frameStart - this->state.lastUpdate);`

marcelstoer, 2021-01-14:
Sure, we always welcome PRs.

hyeok, 2021-01-15:
thanks. next time I will do PR myself.

closed 2021-01-15 by hyeok

marcelstoer, 2021-01-16:
I just noticed that on `master` it already _is_ a `int32_t` thanks to a fix by @geeksville in https://github.com/ThingPulse/esp8266-oled-ssd1306/pull/273. Will be available to all with the upcoming release.