ThingPulse/esp8266-oled-ssd1306 #325: Dstike watch V1

Therealjosephchrzempiec, 2021-01-25:
Hello i have a Dstike watch i got from ebay [https://www.ebay.com/itm/383846316645](url) It works and all but its very hard to program for me that is. My skills are in hardware not really software but I'm trying. The watch only contains a esp8266mod and a 1.3" oled looks like it's using a SH1106 library. I been looking around for a library that can work with this screen and Esp8266mod. 

Everyone i came across has errors because most libraries require a avr processor and not a 32 bit processor. There is a delay error that i have no clue how to fix. I came across this library and no compile errors and say it does run on the esp8266. However i can not get the screen to run. I have tried everything and so far   no luck blank screen. I really need help can someone please help me?

Joseph

closed 2021-10-02 by stale[bot]

khairullaizatt, 2023-01-03:
dstike v1 work with sh1106wire library.

Therealjosephchrzempiec, 2023-01-04:
khairullaizatt, Thank you. I'm working on trying to just get the screen and touch working with just simple commands.

khairullaizatt, 2023-01-05:
can you show me how you solder it ? i’m interested