ThingPulse/esp8266-oled-ssd1306 #327: no Display on OLED

ILohs, 2021-02-07:
Hey, 
my OLED with I2C-Support shows nothing at all when using Lib v4.2.0, but the compiler of the Arduino IDE shows errors.
I am using a Wemos with SCL = D1 and SDA = D2.
The U8G2-Lib, on the other hand, works right away.

This warning comes:
C:\Users\ingol\Documents\Arduino\libraries\ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays\src/SSD1306Wire.h:70:5: warning: unused parameter '_i2cBus' [-Wunused-parameter]
     SSD1306Wire(uint8_t _address, int _sda = -1, int _scl = -1, OLEDDISPLAY_GEOMETRY g = GEOMETRY_128_64, HW_I2C _i2cBus = I2C_ONE, int _frequency = 700000) {
     ^

marcelstoer, 2021-02-08:
Don't worry about the warning - it's just that, no error.

Issue sounds like #311.

ILohs, 2021-02-08:
Solved it with remark: // oled.clear(); in setup()

closed 2021-02-08 by marcelstoer

ezplanet, 2021-05-17:
Try to downgrade to version 4.1.0, it should work.

cjkarande, 2021-07-27:
@ezplanet  I am facing the same issue over I2C (No display on OLED 0.96" SSD1306) even with 4.1.0 for NodeMCU (Esp-12E). However it works perfectly fine with "Adafruit SSD1306"

@ILohs Not working even with clear() call in setup

Do i need to further downgrade to an earlier version?

@marcelstoer