ThingPulse/esp8266-oled-ssd1306 #332: FreeRTOS support

AaronChuzb, 2021-03-13:
When using freertos, a thread is created before ui is initialized, and page scrolling becomes like a delay.

marcelstoer, 2021-03-13:
FreeRTOS isn't one of the supported platforms. However, we'd be happy to accept a fix if you get to the bottom of this.

closed 2021-03-13 by marcelstoer

fake-name, 2022-03-20:
ESP32 is one of the supported platforms. 

The ESP32 IDF (and *all* of their networking features) depend on FreeRTOS. 

So yes, it's one of your supported platforms, even if you don't know it.

marcelstoer, 2022-03-20:
@fake-name your observation is of course correct, thanks. By "not supported platform" I didn't necessarily mean that the lib does not work but more that we are not actively testing against FreeRTOS (outside ESP32).