ThingPulse/esp8266-oled-ssd1306 #339: IMHO you may rename this good lib in esp-oled-ssd1306

rin67630, 2021-04-09:
it works perfectly on the ESP32 as well

Rotzbua, 2021-04-09:
Renaming a git repository is danger zone, because it breaks links automated scripts and many more. Btw the name has been changed for a while in the library.json

marcelstoer, 2021-04-09:
This repository has a long history. At its inception the world didn't have an ESP32 nor did ThingPulse Ltd. exist. Sure, in hindsight it wasn't a good idea to embed the MCU name in the library/repository name. Best we can do is to take this experience into account for future naming challenges.
Btw, even 'esp-oled-ssd1306' wouldn't be appropriate now as the library works on ESP8266, ESP32, and Mbed-OS. 

> the name has been changed for a while in the library.json

Indeed, and that broke some backwards compatibility, too. Library managers aren't very well prepared for such moves.

closed 2021-04-09 by marcelstoer