ThingPulse/esp8266-oled-ssd1306 #343: How to draw curillic or utf 8

lancer-1, 2021-04-25:
is there support for utf-8 text printing and where i can take cyrillic fonts?

marcelstoer, 2021-04-25:
@lorol didn't you play with Cyrillic fonts in your fork?

marcelstoer, 2021-04-25:
In #341 @Shdwdrgn  presented a new font editor. Maybe a way forward would be to start supporting Cyrillic fonts in the various font editors.

Shdwdrgn, 2021-04-25:
From what I've seen, the format of the font files do support more than the base 255 characters so we could technically build a file with as many characters as desired.  I think the real issue is that there's no way to compress the unused characters, so each character will always use a minimum of 4 bytes.  If you're trying to support a full 16-bit character range then you'll be using a minimum of 256k of storage just for a single font (not including the font definitions themselves).  Any efforts to use the upper range need to begin with a way to reduce this to a manageable size.

On the other hand if you can work within the 255 characters of the ANSI set then there shouldn't be any problem.  Since I'm not really clear on what constitutes Cyrillic fonts outside of the ANSI set I don't know if this helps.

lancer-1, 2021-04-26:
@Shdwdrgn thanks a lot! But needed cyrillic symbols are in range 1 to 1111 and its not a problem to use all this, so i think i can rewrite draw function to get more, or not?

lorol, 2021-04-26:
> 
> 
> @lorol didn't you play with Cyrillic fonts in your fork?

Yes I did. Let me check if what I can retrieve. I was long ago.

lorol, 2021-04-26:
https://github.com/lorol/esp8266-oled-ssd1306
It is basically a font header w/ non-ascii letters added, small tweak in OLEDDisplay.cpp code, on the sketch - call utf8ascii() and use UTF-8 on your C source code files so you can write directly the non-ascii strings in it.

lancer-1, 2021-04-26:
@lorol Thanks a lot!

closed 2021-04-26 by lancer-1

marcelstoer, 2021-04-26:
@lorol thanks, a PR would be cool we I totally understand if you do not get around to it.