ThingPulse/esp8266-oled-ssd1306 #345: Upgrade to library version 4.2.0 locks SSH1106 display - all good with version 4.1.0

ezplanet, 2021-05-17:
**Describe the bug**
Run a SSD1306SimpleDemo on a Wemos D1 with SSH1103 display connected to pin D2 (SDA) and D3 (SCL)

**To Reproduce**
Steps to reproduce the behavior:
Change the following:
```c
// #include "SSD1306Wire.h"        // legacy: #include "SSD1306.h"
#include "SH1106Wire.h"   // legacy: #include "SH1106.h"
```

Add the following:
```c
// SSD1306Wire display(0x3c, SDA, SCL);   // ADDRESS, SDA, SCL  -  SDA and SCL usually populate automatically based on your board's pins_arduino.h
SH1106Wire display(0x3c, D2, D3);
```

**Expected behavior**
With library version 4.1.0 everything works fine
After upgrading to version 4.2.0 the screen remains blank and the sketch program freezes
A downgrade back to 4.1.0 restores functionality.
Tried also starting from scratch with the demo provided with Library version 4.2.0

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - Library: [4.2.0]
 - Platform [Wemos D1 + SH1106 128x64 OLED]

**Additional context**
Add any other context about the problem here.

marcelstoer, 2021-05-24:
Sorry, don't have the hardware to test. Assuming the issue is with `src/SH1106Wire.h` which of the 4.1->4.2 changes might cause this: https://github.com/ThingPulse/esp8266-oled-ssd1306/compare/4.1.0...4.2.0#diff-db4440a6b407fac3f22ed405f2f631933bb34cfc22472845d296cb2c88b9ceac ???

nobody1365, 2022-04-05:
Upgrading to 4.3 creates a directory in the libraries folder with some changes in lower and upper case name.
This results in "the old directory is still there"
Compiling that creates a non working dispalay.
to resolve: go to the libraries folder and manually delete the old directory.

bdryanovski, 2022-08-24:
I'm not sure why - but it looks like this lines below are the root of my issues.

```c
// SH1106Wire.h
#if !defined(ARDUINO_ARCH_ESP32) && !defined(ARDUINO_ARCH8266)
      _wire->begin();
#else
      // On ESP32 arduino, -1 means 'don't change pins', someone else has called begin for us.
      if(this->_sda != -1)
        _wire->begin(this->_sda, this->_scl);
#endif 
```

It seems at least for me that I'm into the first section of the `if` block so in my case, I don't call `_wire->begin(...)` so my screen never start.

I'm testing it on `d-duino-b v3` with set board name as `nodemcu v0.9` so my headers may not be super fine ...

AlexeyMal, 2023-07-21:
Dear Bozhidar Dryanovski, thanks for your investigation! 
I confirm the issue and the cause stated in the previous post. 
To fix the problem I commented out the 
  #if ... block with _wire->begin();
to get the command 
  _wire->begin(this->_sda, this->_scl); 
executed on my ESP8266 configured as SDA_PIN = D1 and SDC_PIN = D3. 
With this hotfix my SH1106 display is working again. Without this fix the display never gets initialized and remains black.

Dear Marcel Stör, please fix this issue in the next release!
Thank you!

sprior, 2023-12-21:
I just got bit by this same issue.  Here is my fairly minimal testcase:

platformio.ini:
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
monitor_speed = 115200
lib_deps = 
	Wire
	thingpulse/ESP8266 and ESP32 OLED driver for SSD1306 displays@4.1.0

main.cpp:
#include <Arduino.h>

#include "SH1106.h"

#define I2C_ADDR 0x3c
#define OLED_SDA D3
#define OLED_SCL D5

SH1106 display(I2C_ADDR, OLED_SDA, OLED_SCL);

void setup() {
  Serial.begin(115200);
  while(!Serial) {} // Wait
  Serial.println();

  Serial.println("SDA pin: " + String(OLED_SDA));
  Serial.println("SCL pin: " + String(OLED_SCL));
  display.init();
  display.flipScreenVertically();

  display.drawString(0, 0, "Hello1 World!");
  display.display();
}

void loop() {
}

It works with library version 4.1.0 but fails if I move to 4.2.0 or later.
I'd love to see this fixed, thanks.

marcelstoer, 2023-12-21:
@bdryanovski @sprior @AlexeyMal if the process ends up in the wrong if/else branch it likely means that your board definition does not include the necessary `ARDUINO_ARCH***` defines. Or...wait, shouldn't it be `ARDUINO_ARCH_ESP8266` rather than `ARDUINO_ARCH8266`? I guess that one slipped through when @benoitm974 added support for 2nd HW I2C in #292.

benoitm974, 2023-12-21:
Indeed good catch... Strange we didn't have this issue earlier... 8266 ?
it should say ARDUINO_ARCH_ESP8266

marcelstoer, 2023-12-21:
> Strange we didn't have this issue earlier

Well, this issues is 2.5y old. I lack the hardware to test and it didn't click until I revisited the code sample that @bdryanovski posted above.

closed 2023-12-21 by marcelstoer

benoitm974, 2023-12-21:
@marcelstoer sorry my remark was not a complaints at all, I just surprise there is not so much 8266 users.

marcelstoer, 2023-12-21:
@benoitm974 All good 😄 I didn't take it as complaint. These days the SH1106 + ESP8266 combination is maybe becoming rare.

sprior, 2023-12-21:
Thanks, what's the process to get this released and known to the PlatformIO library registry?

marcelstoer, 2023-12-21:
@sprior It's a fairly simple process that doesn't take much time but I didn't get around to it during the day.

- [bump the version](https://github.com/ThingPulse/esp8266-oled-ssd1306/commit/53a688e5124bc92a11a38d7d17b7c7184ca99d5b) in the config files
- create and push [a tag](https://github.com/ThingPulse/esp8266-oled-ssd1306/tags)
- [cut a release](https://github.com/ThingPulse/esp8266-oled-ssd1306/releases/tag/4.4.1) for said tag
- for the PlatformIO indexer to update [the listing](https://registry.platformio.org/libraries/thingpulse/ESP8266%20and%20ESP32%20OLED%20driver%20for%20SSD1306%20displays), same for [the Arduino libraries](https://www.arduinolibraries.info/libraries/esp8266-and-esp32-oled-driver-for-ssd1306-displays)

sprior, 2023-12-21:
@marcelstoer  ok thanks, I look forward to seeing it when you get to it.

sprior, 2023-12-23:
Thanks @marcelstoer , I confirmed that 4.4.1 fixes the issue.

commit d27d4a5a (2023-12-21): Fix invalid ARDUINO_ARCH_ define

Fixes #345