ThingPulse/esp8266-oled-ssd1306 #354: use the 1.3-inch sh1106 display screen with example， It reports an error

yunxiaguang, 2021-09-17:
**Describe the bug**
I use the 1.3-inch sh1106 display screen, and then compile it with the routine in the example. The error is as follows:
```
Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (3.2.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.2.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
src\main.cpp:63:34: error: no matching function for call to 'SH1106Spi::SH1106Spi(const uint8_t&, const uint8_t&)'
   63 |  SH1106Spi         display(D0, D2);
      |                                  ^
In file included from src\main.cpp:44:
C:\Users\xin\.platformio\packages\framework-arduinoespressif8266\libraries\esp8266-oled-ssd1306-master\src/SH1106SPi.h:43:5: note: candidate: 'SH1106Spi::SH1106Spi(uint8_t, uint8_t, uint8_t, OLEDDISPLAY_GEOMETRY)'
   43 |     SH1106Spi(uint8_t _rst, uint8_t _dc, uint8_t _cs, OLEDDISPLAY_GEOMETRY g = GEOMETRY_128_64) {
      |     ^~~~~~~~~
C:\Users\xin\.platformio\packages\framework-arduinoespressif8266\libraries\esp8266-oled-ssd1306-master\src/SH1106SPi.h:43:5: note:   candidate expects 4 arguments, 2 provided
C:\Users\xin\.platformio\packages\framework-arduinoespressif8266\libraries\esp8266-oled-ssd1306-master\src/SH1106SPi.h:37:7: note: candidate: 'constexpr SH1106Spi::SH1106Spi(const SH1106Spi&)'
   37 | class SH1106Spi : public OLEDDisplay {
      |       ^~~~~~~~~
C:\Users\xin\.platformio\packages\framework-arduinoespressif8266\libraries\esp8266-oled-ssd1306-master\src/SH1106SPi.h:37:7: note:   candidate expects 1 argument, 2 provided
C:\Users\xin\.platformio\packages\framework-arduinoespressif8266\libraries\esp8266-oled-ssd1306-master\src/SH1106SPi.h:37:7: note: candidate: 'constexpr SH1106Spi::SH1106Spi(SH1106Spi&&)'
C:\Users\xin\.platformio\packages\framework-arduinoespressif8266\libraries\esp8266-oled-ssd1306-master\src/SH1106SPi.h:37:7: note:   candidate expects 1 argument, 2 provided
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
```

Then I found the problem in line 32 and changed it to _**SH1106Spi   display(D0, D2，D8);**_
Then it succeeded

```

1. // Include the correct display library
2. // For a connection via I2C using Wire include
3. //#include <Wire.h>  // Only needed for Arduino 1.6.5 and earlier
4. //#include "SSD1306Wire.h" // legacy include: `#include "SSD1306.h"`
5. // or #include "SH1106Wire.h", legacy include: `#include "SH1106.h"`
6. // For a connection via I2C using brzo_i2c (must be installed) include
7. // #include <brzo_i2c.h> // Only needed for Arduino 1.6.5 and earlier
8. // #include "SSD1306Brzo.h"
9. // #include "SH1106Brzo.h"
10. // For a connection via SPI include
11. #include <SPI.h> // Only needed for Arduino 1.6.5 and earlier
12. // #include "SSD1306Spi.h"
13. #include "SH1106SPi.h"
14. 
15. // Include the UI lib
16. #include "OLEDDisplayUi.h"
17. 
18. // Include custom images
19. 
20. #include "images.h"
21. 
22. // Use the corresponding display class:
23. 
24. // Initialize the OLED display using SPI
25. // D5 -> CLK
26. // D7 -> MOSI (DOUT)
27. // D0 -> RES
28. // D2 -> DC
29. // D8 -> CS
30. // SSD1306Spi        display(D0, D2, D8);
31. // or
32.  SH1106Spi   display(D0, D2);

```
第一次反馈问题，而且还是英文，写得不好多多包涵。

The first time I fed back the question, and it was still in English. If I didn't write well, please forgive me.

thijstriemstra, 2021-09-19:
> Then it succeeded

Cool, so is there any issue to report?

marcelstoer, 2021-09-20:
Think there are two issues actually.

1. The commented SH1106Spi lines in the examples e.g. as seen here https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/216812cbd803151fae930b1328520251198ab216/examples/SSD1306ClockDemo/SSD1306ClockDemo.ino#L63 defines only two out of the three required parameters.
2. The third parameter (`_cs`) is never actually used? https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/216812cbd803151fae930b1328520251198ab216/src/SH1106Spi.h#L43-L48

yunxiaguang, 2021-09-23:
YES,That's the problem I want to reflect。

closed 2022-04-30 by stale[bot]