ThingPulse/esp8266-oled-ssd1306 #355: ESP32S2 - I2C Conflict with Other Libraries

lcsjunior, 2021-09-26:
I have recently been using an [ESP32 S2 LOLIN S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html).
I used the [sensirion arduino-sht](https://github.com/Sensirion/arduino-sht) library for sht30 i2c together with Adafruit_SSD1306, but I prefer this ThingPulse OLED SSD1306.
It turns out that ThingPulse OLED SSD1306 makes my sht30 not startable.

See:
- Adafruit_SSD1306 and arduino-sht work well together.
- The ThingPulse OLED SSD1306 examples work fine alone.
- Sensirion arduino-sht works fine only.

Code:
...
#define D1_SCL  35
#define D2_SDA  33
...
SHTSensor sht;
SH1106Wire display(0x3c, D2_SDA, D1_SCL);.
...
void setup() {
  Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }
  sht.setAccuracy(SHTSensor::SHT_ACCURACY_MEDIUM);

  display.init();
  display.flipScreenVertically();
  display.setFont(ArialMT_Plain_10);
...

marcelstoer, 2021-09-26:
I don't have the hardware to test. Hence, I can only do a dry-run analysis.

Did you try setting the SHT address/type explicitly e.g. `SHTSensor sht(SHTSensor::SHT3X);` as per https://github.com/Sensirion/arduino-sht/blob/master/SHTSensor.h#L49? Also, you don't need to call `Wire.begin()` for this library - we do it for you. Just run the display initialization _before_ the `sht.init()`.

lcsjunior, 2021-09-26:
@marcelstoer I haven't had success yet. By the way, I have been having a lot of headache with arduino libraries when I use ESP32-S2. OneWire is one of them.

// ------------------- SHT OK

SSD1306Wire display(0x3c, D2_SDA, D1_SCL);
SHTSensor sht(SHTSensor::SHT3X_ALT);

void setup() {
  Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  // display.init();

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }

// ------------------- SHT failed

SSD1306Wire display(0x3c, D2_SDA, D1_SCL);
SHTSensor sht(SHTSensor::SHT3X_ALT);

void setup() {
  Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  display.init();

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }

// ------------------- SHT failed

SSD1306Wire display(0x3c, D2_SDA, D1_SCL);
SHTSensor sht(SHTSensor::SHT3X_ALT);

void setup() {
  // Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  // display.init();

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }

// ------------------- SHT failed

SSD1306Wire display(0x3c, D2_SDA, D1_SCL);
SHTSensor sht(SHTSensor::SHT3X_ALT);

void setup() {
  // Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  display.init();

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }

// ------------------- SHT failed

SSD1306Wire *display;
SHTSensor sht(SHTSensor::SHT3X_ALT);

void setup() {
  // Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  SSD1306Wire dp(0x3c, D2_SDA, D1_SCL);
  display = &dp;
  display->init();

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }

// ------------------- SHT failed

SSD1306Wire *display;
SHTSensor sht(SHTSensor::SHT3X_ALT);

void setup() {
  Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  SSD1306Wire dp(0x3c, D2_SDA, D1_SCL);
  display = &dp;
  display->init();

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }

// ------------------- SHT failed

SSD1306Wire *display;
SHTSensor sht(SHTSensor::SHT3X_ALT);

void setup() {
  Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }
  sht.setAccuracy(SHTSensor::SHT_ACCURACY_MEDIUM);

  SSD1306Wire dp(0x3c, D2_SDA, D1_SCL);
  display = &dp;
  display->init();

lcsjunior, 2021-09-26:
@marcelstoer I've had success with U8g2 lib and arduino-sht.

SHTSensor sht(SHTSensor::SHT3X_ALT);
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);

void setup() {
  Wire.begin(D2_SDA, D1_SCL);
  Serial.begin(115200);

  if (!sht.init()) {
    Serial.println(F("SHT initialization failed"));
  }
  sht.setAccuracy(SHTSensor::SHT_ACCURACY_MEDIUM);

  u8g2.begin();

But, SHT doesn't work if I enter SDA and SDL in U8G2.

U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, **D1_SCL, D2_SDA**, U8X8_PIN_NONE);

illosan, 2021-12-30:
I solved the incompatibility problems with the AM2320 temperature sensor library initializing the oled display in the following way:

SSD1306Wire d1306 (0x3c, 21, 22, GEOMETRY_128_32, I2C_ONE, -1);

phpYida, 2022-01-03:
my project was I2C Conflict with Other Libraries too;
my device is esp8266;
I solved in this way

in setup method code:

bme.begin();  //another i2c sensor 
 Wire.begin(D3, D4);//my ssd1306 connect in d3 ,d4 port
 display.init();
....

then

 in loop method code:
Wire.begin(SDA, SCL); //set your sensor  port 
temp = bme.readTemperature() //just read my bme sensor data
display.clear();
Wire.begin(D3, D4);//my ssd1306 connect in d3 ,d4 port
display.setTextAlignment(TEXT_ALIGN_LEFT);
display.setFont(ArialMT_Plain_16);
display.drawString(0, 0, "temp:" + String(temp)  + "°C");
display.display();

hope can help you

lcsjunior, 2022-01-03:
Actually, I've been used e-paper display. But, I suspect that this problem is pins_arduino.h in the esp32s2 dev module on the esp32 arduino library. Because, the S2 mini pins in the schema are differents than in the pins_arduino.h.

closed 2022-01-03 by lcsjunior