ThingPulse/esp8266-oled-ssd1306 #369: display.drawString() cannot utilize the first rows of the display

flyvert, 2022-04-09:
I've already made a post here https://forum.arduino.cc/t/ssd1306wire-font-alignment-problem/978025
I then realized I can post an issue here on GitHub.

Library version 4.3.0. Arduino 1.8.13. macOS 12.2 (21D49). Generic ESP8266.

The drawString() seem unable to draw text at the first pixel rows of the display. I have a thermometer project where I want to maximize two temperature readings on a 128x64 display - I have experimented with creating a large font and is currently using a 34 pixel version of Sans Serif which looks nice on the pixels I've been able to utilize, however, there is some space left unused.

A call to drawstring(0,0...) leaves several pixels (the more the larger the font size used) unused at the top of the display. Using a smaller font moves text upwards

Here below two examples giving a hands on view of the problem.

1) Notice the smaller 10 pixel font starts above the 34 larger font, both drawn at row 0 - still a drawLine on row 0 is painted above both texts.

display.setFont(ArialMT_Plain_10);
display.drawString(95,0, "toprow");
display.drawLine(0,0,128,0);

2) Notice a rectangle on the last pixels of the screen. The first text row (drawn at 0,0) is clearly below the top while the second text row can be put exactly on the bottom row (I can't recall if text was starting at pixel 32, however, I am able to tweak the text to draw at the bottom pixel row).

display.setFont(SansSerif_Plain_34);
display.drawString(0, 0, "12.3°");
display.drawString(0, 32, "12.3°");
display.drawRect(0,0,128,64);

[image]
[image]

andreaslanz, 2022-04-10:
>  I have experimented with creating a large font and is currently using a 34 pixel version of Sans Serif which looks nice on the pixels I've been able to utilize, however, there is some space left unused.

Hi
that could be the problem. can you post this font
Do you have it with:
`display.drawString(0, -1, "12.3°");`
tried?

flyvert, 2022-04-10:
Thanks for your response.

Supplying a negative offset didn't work for me.

I've attached a ZIP of the _OLEDDisplayFonts.h_ I'm using - the only difference from the original is the Sans Serif 34 added just after the Arial 24. Font was compiled on macOS using the web tool suggested in the library's README.md

[OLEDDisplayFonts.h.zip](https://github.com/ThingPulse/esp8266-oled-ssd1306/files/8458604/OLEDDisplayFonts.h.zip)

flyvert, 2022-04-10:
> Hi that could be the problem. can you post this font Do you have it with: `display.drawString(0, -1, "12.3°");` tried?

Even without my custom 34 pixel font, I believe you can reproduce the issue using the "toprow" example I posted - the Arial 10 text seem to start at the third pixel row compared to the horizontal line I drew above it using display.drawLine(0,0,128,0);
[image]

andreaslanz, 2022-04-11:
Yes you are right, I tested the following on a heltec_wifi_lora_32_V2:

```
     // FontDemo1
   
     display. drawLine(0,0,128,0); // only to mark the top of the display
     display.setFont(ArialMT_Plain_16);
     display. drawString(0, -5, "Hello world"); // Must go to the Top
     display.display();

```
But there is a 2 pixel gap to the top edge

[image]

flyvert, 2022-04-11:
> Yes you are right, I tested the following on a heltec_wifi_lora_32_V2:
>
> ```
>      // FontDemo1
>
>      display. drawLine(0,0,128,0); // only to mark the top of the display [...]

Thanks for repeating the problem.

I believe a large font will add more “dead” pixels, I guess 5-6 at 34pt.

Can I expect a fix soon?

Do you know a workaround or tweak? I made some lame attempts to adjust the Y (height) setting in fonts.h that did not have any effect.

andreaslanz, 2022-04-11:
> Can I expect a fix soon?

hmm... currently I'm not involved in the development of the library. I'm just an "user" ;-)

Maybe this is not a bug but a feature. Text needs to be some distance from the edge for '**aesthetic**' reasons. I do not know.

Or there is a hidden parameter to switch this off.

flyvert, 2022-04-11:
> > Can I expect a fix soon?
> 
> hmm... currently I'm not involved in the development of the library. I'm just an "user" ;-)
> 
> Maybe this is not a bug but a feature. Text needs to be some distance from the edge for '**aesthetic**' reasons. I do not know.
> 
> Or there is a hidden parameter to switch this off.

Since I can draw text to the far bottom, I expect to be able to draw to far top as well.

I'm in no jam, just not able to use nearly 10% of the height of the display for two rows of a 34pt font. If this bug gets corrected I guess I can squeeze some more point into the font tool until the two rows of temperature readings fully occupy the full 64 pixels of the display (with a couple of blank rows in between).

I hope some dev or "maintenance squad" can add their thoughts.

b73f9, 2022-07-09:
The tool that generates the fonts seems to be the thing at fault here, I was able to get the default arial 10 font to go all the way up to the top of the screen by shifting the font data 3 bits up. 
This might be because of some of the special chars at the end of the ascii table which is mostly useless but does have some data in those top 3 bits, but not sure about the specifics. 

```cpp
const uint8_t ArialMT_Plain_10_fixed[] = {
  0x0A, // Width: 10
  0x0D, // Height: 13
  0x20, // First Char: 32
  0xE0, // Numbers of Chars: 224

  // Jump Table:
  0xFF, 0xFF, 0x00, 0x03,  // 32:65535
  0x00, 0x00, 0x04, 0x03,  // 33:0
  0x00, 0x04, 0x05, 0x04,  // 34:4
  0x00, 0x09, 0x09, 0x06,  // 35:9
  0x00, 0x12, 0x0A, 0x06,  // 36:18
  0x00, 0x1C, 0x10, 0x09,  // 37:28
  0x00, 0x2C, 0x0E, 0x07,  // 38:44
  0x00, 0x3A, 0x01, 0x02,  // 39:58
  0x00, 0x3B, 0x06, 0x03,  // 40:59
  0x00, 0x41, 0x06, 0x03,  // 41:65
  0x00, 0x47, 0x05, 0x04,  // 42:71
  0x00, 0x4C, 0x09, 0x06,  // 43:76
  0x00, 0x55, 0x04, 0x03,  // 44:85
  0x00, 0x59, 0x03, 0x03,  // 45:89
  0x00, 0x5C, 0x04, 0x03,  // 46:92
  0x00, 0x60, 0x05, 0x03,  // 47:96
  0x00, 0x65, 0x0A, 0x06,  // 48:101
  0x00, 0x6F, 0x08, 0x06,  // 49:111
  0x00, 0x77, 0x0A, 0x06,  // 50:119
  0x00, 0x81, 0x0A, 0x06,  // 51:129
  0x00, 0x8B, 0x0B, 0x06,  // 52:139
  0x00, 0x96, 0x0A, 0x06,  // 53:150
  0x00, 0xA0, 0x0A, 0x06,  // 54:160
  0x00, 0xAA, 0x09, 0x06,  // 55:170
  0x00, 0xB3, 0x0A, 0x06,  // 56:179
  0x00, 0xBD, 0x0A, 0x06,  // 57:189
  0x00, 0xC7, 0x04, 0x03,  // 58:199
  0x00, 0xCB, 0x04, 0x03,  // 59:203
  0x00, 0xCF, 0x0A, 0x06,  // 60:207
  0x00, 0xD9, 0x09, 0x06,  // 61:217
  0x00, 0xE2, 0x09, 0x06,  // 62:226
  0x00, 0xEB, 0x0B, 0x06,  // 63:235
  0x00, 0xF6, 0x14, 0x0A,  // 64:246
  0x01, 0x0A, 0x0E, 0x07,  // 65:266
  0x01, 0x18, 0x0C, 0x07,  // 66:280
  0x01, 0x24, 0x0C, 0x07,  // 67:292
  0x01, 0x30, 0x0B, 0x07,  // 68:304
  0x01, 0x3B, 0x0C, 0x07,  // 69:315
  0x01, 0x47, 0x09, 0x06,  // 70:327
  0x01, 0x50, 0x0D, 0x08,  // 71:336
  0x01, 0x5D, 0x0C, 0x07,  // 72:349
  0x01, 0x69, 0x04, 0x03,  // 73:361
  0x01, 0x6D, 0x08, 0x05,  // 74:365
  0x01, 0x75, 0x0E, 0x07,  // 75:373
  0x01, 0x83, 0x0C, 0x06,  // 76:387
  0x01, 0x8F, 0x10, 0x08,  // 77:399
  0x01, 0x9F, 0x0C, 0x07,  // 78:415
  0x01, 0xAB, 0x0E, 0x08,  // 79:427
  0x01, 0xB9, 0x0B, 0x07,  // 80:441
  0x01, 0xC4, 0x0E, 0x08,  // 81:452
  0x01, 0xD2, 0x0C, 0x07,  // 82:466
  0x01, 0xDE, 0x0C, 0x07,  // 83:478
  0x01, 0xEA, 0x0B, 0x06,  // 84:490
  0x01, 0xF5, 0x0C, 0x07,  // 85:501
  0x02, 0x01, 0x0D, 0x07,  // 86:513
  0x02, 0x0E, 0x11, 0x09,  // 87:526
  0x02, 0x1F, 0x0E, 0x07,  // 88:543
  0x02, 0x2D, 0x0D, 0x07,  // 89:557
  0x02, 0x3A, 0x0C, 0x06,  // 90:570
[... 250 lines cut ...]
  0x04, 0x00, 0x7f, 0x00, 0x44, 0x00, // 116
  0x00, 0x00, 0x3c, 0x00, 0x40, 0x00, 0x40, 0x00, 0x7c, 0x00, // 117
  0x04, 0x00, 0x38, 0x00, 0x40, 0x00, 0x38, 0x00, 0x04, // 118
  0x3c, 0x00, 0x40, 0x00, 0x38, 0x00, 0x04, 0x00, 0x38, 0x00, 0x40, 0x00, 0x3c, 0x00, // 119
  0x44, 0x00, 0x28, 0x00, 0x10, 0x00, 0x28, 0x00, 0x44, 0x00, // 120
  0x04, 0x00, 0x38, 0x01, 0xc0, 0x00, 0x38, 0x00, 0x04, // 121
  0x44, 0x00, 0x64, 0x00, 0x54, 0x00, 0x4c, 0x00, 0x44, 0x00, // 122
  0x10, 0x00, 0xef, 0x01, 0x01, 0x01, // 123
  0x00, 0x00, 0xff, 0x01, // 124
  0x01, 0x01, 0xef, 0x01, 0x10, // 125
  0x18, 0x00, 0x08, 0x00, 0x18, 0x00, 0x10, 0x00, 0x18, // 126
  0x00, 0x00, 0xf4, 0x01, // 161
  0x00, 0x00, 0x38, 0x00, 0xf4, 0x01, 0x4f, 0x00, 0x28, 0x00, // 162
  0x48, 0x00, 0x6e, 0x00, 0x59, 0x00, 0x49, 0x00, 0x41, 0x00, 0x42, 0x00, // 163
  0x00, 0x00, 0x3c, 0x00, 0x24, 0x00, 0x24, 0x00, 0x3c, 0x00, // 164
  0x29, 0x00, 0x2e, 0x00, 0x78, 0x00, 0x2e, 0x00, 0x29, 0x00, // 165
  0x00, 0x00, 0xe7, 0x01, // 166
  0x9a, 0x00, 0x25, 0x01, 0x29, 0x01, 0x49, 0x01, 0xb2, 0x00, // 167
  0x01, 0x00, 0x00, 0x00, 0x01, // 168
  0x1c, 0x00, 0x22, 0x00, 0x49, 0x00, 0x55, 0x00, 0x55, 0x00, 0x22, 0x00, 0x1c, // 169
  0x0d, 0x00, 0x0d, 0x00, 0x0d, 0x00, 0x0f, // 170
  0x00, 0x00, 0x30, 0x00, 0x48, 0x00, 0x30, 0x00, 0x48, 0x00, // 171
  0x04, 0x00, 0x04, 0x00, 0x04, 0x00, 0x04, 0x00, 0x1c, // 172
  0x10, 0x00, 0x10// 173
};```

marcelstoer, 2022-07-09:
@b73f9 interesting observation. Any chance you can locate the potential issue in the tool's code at https://github.com/squix78/esp8266-oled-ssd1306-font-converter?

b73f9, 2022-07-09:
Hm, sorry, don't have an easy way to run that / experiment with it. 
I'm guessing this specific bit might be at fault 
https://github.com/squix78/esp8266-oled-ssd1306-font-converter/blob/a1428ba4f0ec3ae09a360c8413f5b3f7fbd849d8/src/main/java/ch/squix/esp8266/fontconverter/rest/FontConverterV3.java#L204= 
(basically I'd guess this is displaying the characters with too much space above them, which then gets converted into the empty bits)

MrSniffer, 2022-10-28:
I had a symptom which I think is related to the same bug. I am using ArialMT_Plain_10 and used to be able to use display.drawString(0,10*n,screen[n]);} to draw 6 lines of text, n=0...5.
With recent versions of this library, line 5 is split, half at the bottom of the display and half at the top. The other lines are visible, but everything is "rotated" down 1/2 line.