ThingPulse/esp8266-oled-ssd1306 #370: Can not use the "Wire1",as the secend i2c

voidlaobai, 2022-05-08:
**Describe the bug**
SSD1306 display(0x3c, oled_SDA, oled_SCL,GEOMETRY_128_64,I2C_TWO);//0x3c, SDA, SCL
i did seclect the I2C_TWO,but it don't work 

and there is a only seclection :HW_I2C,a soft I2C would be better,SF_I2C
(in oleddisplay.h 146)
just for esp device compatibility,this wouldn't be big problem.

(thank u for help,and the lib really give a lot of help)specilly esp32c3 can't use u8g2

voidlaobai, 2022-05-08:
i know.
esp32-c3 only has ONE i2c channle

voidlaobai, 2022-05-08:
esp32-c3 workout by this way,if u really need use two i2c
Wire.begin(oled_SDA,oled_SCL);
  display.drawString(0,0,"Hello World from ESP32!");
  display.display();
  Wire.end();  
  delay(1000);
  Serial.println("running");
  Wire.begin(ads1115_SDA,ads1115_SCL);
  ads_str = ads.computeVolts(ads.readADC_SingleEnded(0));
  Serial.println( ads_str);
  Serial.println(double (ads.readADC_SingleEnded(0)*4.096/32768) );
  Wire.end();

HACKER-3000, 2022-07-28:
i suggest you close the issue if you have solved it yourself

closed 2022-07-28 by marcelstoer