ThingPulse/esp8266-oled-ssd1306 #378: The bitmap drawing is not compatible with the img2cpp websites. The data needs to be fliped every 8bits.

sonictl, 2022-12-15:
**Describe the bug**

I'm using https://javl.github.io/image2cpp/ for converting the images into xbm data.
However the data generated cannot be displayed correctly. xbm image is flipped every 8 bit.

ESP8266
0.91 inch 128x32 ssd1306 oled.

This incompatible issue can be rebuilt on web tools:
http://dot2pic.com/
https://diyusthad.com/image2cpp
https://javl.github.io/image2cpp/

They are the most easy to use web tools for xbm binary data generation that I can find.

khairullaizatt, 2022-12-30:
ive been struggling with the result which is display not compatible. does flipped every 8 bit. can someone show me the right method ?

leonardojc, 2023-06-28:
you must use GIMP to export the images in 1-bit XBM format.
Works perfectly

eitherway, 2023-07-11:
@sonictl
You don't need to use Gimp. The https://javl.github.io/image2cpp/ works fine, if you adjust the output settings:
- No Image Rotation
- Draw Mode: Horizontal (1bit)
- Swap bits in byte: swap <-- This is the important setting.

Thanks for the issue! Without your information, I would have had to download Gimp.

marcelstoer, 2023-07-11:
@eitherway thanks for the comment. Would you be willing to help support this project by creating a PR which extends the documentation in the README?

closed 2023-07-11 by marcelstoer

eitherway, 2023-07-12:
@marcelstoer I opened a pull request