ThingPulse/esp8266-oled-ssd1306 #381: Unused variable warning on ESP8266

nyanpasu64, 2023-03-15:
**Describe the bug**
On platforms other than ESP32, I get an unused variable warning in the `SSD1306Wire` constructor, since only the 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Do this '....'
3. Do that '....'
4. See error

**Sample code**
Provide a [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) below.
```c
# your code goes here
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - Library: [e.g. 4.2]
 - Platform [e.g. ESP266 Arduino Core 2.5.x]

**Additional context**
Add any other context about the problem here.

marcelstoer, 2023-03-25:
@benoitm974 what is your take on this? I ask for your opinion as you contributed the code in question back then.

nyanpasu64, 2023-03-25:
I actually feel the "unused arguments" warning is too noisy and turn it off in my personal projects, but it's difficult to set custom compiler flags on Arduino IDE v1, and reducing the warning level turns off other useful warnings as well. In this case I think `(void)_i2cBus` isn't too ugly of a change to make.

In any case, should the _i2cBus argument even exist outside of ESP32? Though removing the argument on some platforms can break code on a platform-dependent basis, which is even worse than leaving it in.

Should the HW_I2C type only define I2C_TWO on ESP32, and otherwise be a one-value enum to prevent users from passing in I2C_TWO on ESP8266 which is silently ignored? Or is it better to leave I2C_TWO in?

benoitm974, 2023-03-26:
since this code is executed at init only why don't secure it by adding a test that someone is not trying to use a second wire on another ESP32 platform ?

(_i2cBus!=I2C_ONE)? ASSERT (not supported 2nd Wire blablabal... ?

Benoit