ThingPulse/esp8266-oled-ssd1306 #387: SSD1306TwoScreenDemo ESP32 doesn't work

c3Vw, 2023-06-29:
Hello,

I believe 2 screen demo doesn't work properly.
Only 1 screen works at a time.
I'm developing on ESP WROOM 32 .

Below is my setup:
`SSD1306Wire  display(0x3c, 26, 27, GEOMETRY_128_32);
SSD1306Wire  display2(0x3c, 19, 18, GEOMETRY_128_32);`

Issue is that only display works correctly, while display2 is completly blank. 
If I comment out all display object functions display2 works as expected.

Any ideas what's wrong?

lafouk, 2023-11-15:
Same issue there, it used to work. Something probably change on the dependencies level, maybe Wire.h ?
Just tried to use a legacy version of Wire on my side, and nothing's better for now.

If someone had some success on this topic, I'd be glad to hear about it.

FriqueNFraque, 2024-01-13:
bump. tried with an ESP32S1 and not working.

michelgerritse, 2025-05-12:
Here's a fix for the problem (tested on ESP32-S3):

```
// Initialize the OLED display using Wire library
SSD1306Wire  display(0x3c, 4, 5, GEOMETRY_128_64, I2C_ONE);
SSD1306Wire  display2(0x3c, 9, 10, GEOMETRY_128_64, I2C_TWO);
```
This will create 2 separate I2C busses. Since both displays have the same address, you can now individually address them.

Alternatively (this is not tested), you can modify/solder the address of the 2nd display to something like 0x3d. This will allow both displays to be available on the same I2C bus.

marcelstoer, 2025-05-13:
@michelgerritse thanks for posting this example. If you work with an ESP32 having two I2C buses (some do, some don't) you generally need consider the corresponding constructor parameter.

Example: https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/master/src/SSD1306Wire.h#L72

Should this be mentioned explicitly in a comment for every constructor in every example sketch? I'd rather not as I wouldn't want to duplicate the API documentation in the examples.

michelgerritse, 2025-05-13:
@marcelstoer It was obvious to me the constructor parameters needed to be changed accordingly. However I think some users expect a compile, upload and run scenario when trying out the examples, which won't work for the TwoScreenDemo example without some modification (they probably got lucky with the other examples). I would just leave it as is because, as you mentioned, duplicating the API documentation is not desired.

BTW: I like the work you did with this library. Will use it in my upcoming "projects".

commit f90368eb (2025-05-14): Add note about i2cBus parameter

Fixes #387
closed 2025-05-14 by marcelstoer