ThingPulse/esp8266-oled-ssd1306 #390: Reset signal

ropg, 2024-02-23:
Some boards, notably the popular Heltec's ESP32_LoRa_v3, have the reset wire hooked up. It might be nice to pull that low for 50 ms or so when connecting. Could do a version of the constructor that has `_rst` an extra parameter (default -1 for no hw reset) between _scl and geometry. There's pros and cons since everything has default values, but given that geometry has its own type, people can't put int in there anyway. Right?

Good idea? Shall I create a PR for that?

ropg, 2024-02-23:
I guess that would break code that does, say, `SSD1306Wire display(0x3c, SDA, SCL, GEOMETRY_128_32);`, so that leaves putting it all the way at the end, or having something like `setResetPin(int _rst)`. Or deciding it out of scope.

marcelstoer, 2024-05-23:
I'd rather not add more constructor parameters. Especially so if they are for auxiliary functionality.

> having something like `setResetPin(int _rst)`

👍