ThingPulse/esp8266-oled-ssd1306 #402: Huge font size - possible?

mister-Monk, 2024-04-02:
I need to show two numbers on the display as large as possible.
Ideally 64 pixels character height.
I chose the **Open_Sans_Hebrew_Condensed** font http://oleddisplay.squix.ch/
I'm guessing size 80 is perfect for this.
But starting from size 63, the screen does not draw the number 4.
And size 80 draws only the edge of the character.
In the editor, I moved all the characters to the left by three pixels.
Now only number 1 is drawn in full.
Only two or three digits need to be displayed.
Not enough buffer size? What to do? 
**ESP8266**, SSD1306 I2C, latest library.
[image]

nickbits1024, 2024-05-21:
the library needs to be modified to work with 16 bit values instead of chars.  notice in the generated font array that some of the values are >255

nickbits1024, 2024-05-21:
If you switch to a <3.0 version and format, it works as-is. 
Also, somewhat related the font creator crashes (500 server error) with the new format (and Adafruit GFX) with sizes >80ish.  <3.0 works.

nickbits1024, 2024-05-21:
If you want to fix the drawing routine for the new version all that needs to be done is change the generated font from char to short/int16_t as well as all the local vars in the drawing routines.

marcelstoer, 2024-05-23:
@nickbits1024 @mister-Monk just pay attention to the overall memory consumption.