ThingPulse/esp8266-oled-ssd1306 #410: My display has gone blank since upgrading to 4.6.0

radioralph, 2024-05-25:
My SSD1306 just stopped working (blank screen) during the development of a project after my IDE installed the 4.6.0 update.  I rolled backed to 4.5.0 and it's working again, but where can I find a list of changes so I can determine what I did wrong and how I can get my display working again with 4.6.0

My environment is ESP32S3 with I2C interface to a 128x64 display.

marcelstoer, 2024-05-25:
The list of changes is in the release notes at https://github.com/ThingPulse/esp8266-oled-ssd1306/releases/tag/4.6.0.

Curious to find out what the issue is in your case.

radioralph, 2024-05-25:
Thank you for your direction to the release notes. Unfortunately, my skill is not strong enough to solve this, however I did notice that in the new release that the **_frequency** variable in line 75 of SSD1306Wire.h has been replaced by **frequency**, ie. without the leading underscore.  For the time being I'm happy to use the previous version.

closed 2024-05-25 by marcelstoer

marcelstoer, 2024-05-25:
Please see if 4.6.1 fixes your issues. If not, please provide a reproducer ([MRE](https://stackoverflow.com/help/minimal-reproducible-example)).

radioralph, 2024-05-25:
Thank you for your message. I am pleased to advise that 4.6.1 works perfectly now. 😊.

 

I appreciate your rapid attention to solving this, Marcel.

 

Kind regards,

Ralph

commit ed1feac8 (2024-05-25): Fix PR #404

Fixes #410, #411