ThingPulse/esp8266-oled-ssd1306 #411: I2C Error with 4.6.0 upgrade

Trebor9691, 2024-05-25:
Today I upgraded the library to version 4.6.0 and immediately things went wrong as I am currently working on a project that uses the library. As another as stated, my screen is also blank however, to add a little more light to the issue I also noticed the following:
1) The issue is happening when initializing the screen and does not allow my code to continue.
2) There is an error output to the Serial Monitor when the problem occurs that states that the I2C pins SDA and SCL are the same.

I have downgraded to 4.5.0 and everything is working again. I hope the additional information is helpful to you.

Sincerely,
~Rob

marcelstoer, 2024-05-25:
I think I found the issue. In #404 from @ClutchplateDude there's a refactoring missing in the `SSD1306Wire` constructor at https://github.com/ThingPulse/esp8266-oled-ssd1306/commit/469f227ea567741f0513c572f90f66c7595c139d#diff-fddf375493ceb756b8da0701d1bb96b6e3bdefdb23046c2ef69f9bd1d73bb3ffR78-R86.

ClutchplateDude, 2024-05-25:
Thanks for fixing that @marcelstoer , no idea how that got through and didn't trigger build failures on my end....

marcelstoer, 2024-05-25:
Fixed with 4.6.1.

closed 2024-05-25 by marcelstoer

marcelstoer, 2024-05-25:
No worries, just bad luck that it's a kind of error that doesn't trigger a CI failure either.