ThingPulse/esp8266-oled-ssd1306 #417: [deprecated] Print functionality now handles buffer management automatically. This is a no-op.

kskenyon, 2025-02-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Do this '....'
3. Do that '....'
4. See error

**Sample code**
Provide a [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) below.
```c
# your code goes here
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - Library: [e.g. 4.2]
 - Platform [e.g. ESP266 Arduino Core 2.5.x]

**Additional context**
Add any other context about the problem here.

marcelstoer, 2025-02-13:
No need to call the no-op functions like `drawLogBuffer(x,y)`, see https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/master/src/OLEDDisplay.cpp#L828-L832

closed 2025-02-13 by marcelstoer

kskenyon, 2025-02-13:
OK, thanks,  I reverted to 4.3.0 and all is well.  Couldn’t find that call in the code.

Kevin

marcelstoer, 2025-02-13:
It's right here https://github.com/Xinyuan-LilyGO/tbeam-helium-mapper/blob/main/main/screen.cpp#L288

kskenyon, 2025-02-13:
> marcelstoer
>  left a comment
> (ThingPulse/esp8266-oled-ssd1306#417)
> It's right here https://github.com/Xinyuan-LilyGO/tbeam-helium-mapper/blob/main/main/screen.cpp#L288

Thanks,  I looked in main.cpp.  Duh.  Is there a reason to use 4.6.1 instead of 4.3.0 for practical purposes?  Seems to be working OK.

Kevin